ClockSelect/myevic #1: Hide modes and bypass

Daveee10, 2016-08-06:
Hello!
First, sorry for my bad english.
I found a bug. If you hide the Ni and Ti modes but the SS not, you cant select Temp SS just you select that befor hide another temp modes. This issue like if hide Ni and SS but Ti not and you want select Ti. This works only when you hide only unselected modes.
Another issue the bypass mode... The calculated wattage not correct... 2.03ohm and 3,75V battery print 37,6W.. But this is 6,9W in real....
Third issue is the board temp in 3rd line... I think this is only positioning problem or string lenght problem. I don't know. I haven't got any time now for checking code, but next week maybe i will check it.

Awesome work bro! Keep calm and make magic with this project ;)

ClockSelect, 2016-08-11:
I see. Having a look.

ClockSelect, 2016-08-11:
Both issues fixed.
Bypass power display came from a misdisassembled function (hex-rays fault, this time) and temp on third line caused a misalignement issue when in Farenheits.

closed 2016-08-11 by ClockSelect

ClockSelect, 2016-08-11:
Thanks for your reports :)