ClockSelect/myevic #2: Hardware version is cleared

TBXin, 2016-08-08:
After uploading your firmware hardware version in the dataflash erased with following bytes FF FF FF FF.
This can "brick" the device after rolling back to the stock firmware, since they are using this data to initialize correct LCD 1306 or 1327.

Also, reset dataflash function does not restore correct HW Ver. 
The only way is to manually edit dataflash and then upload it.

ClockSelect, 2016-08-11:
Fixed.

The DataFlash CRC was not correctly computed, since the CRC_MODULE clock was not enabled. So the OFW considered all DF pages as invalid.

closed 2016-08-11 by ClockSelect

ClockSelect, 2016-08-11:
(Anyway, that's strange: uneless everything has been cleared in the dataflash, the hardware version should always be recovered, even if CRC is bad).

ClockSelect, 2016-08-11:
Thanks for your reports!

TBXin, 2016-08-11:
Good job! Now it's work properly :+1:

slavyk, 2016-08-12:
Great work but is possible to restore original HW version?! My eVic was HW1.00...

TBXin, 2016-08-12:
@slavyk i know one way:
1. Read dataflash from your device to PC via NFirmwareEditor or python-evic;
2. Manually edit file via HEX-editor of your choice;
3. Write dataflash to your device;

Explaining 2nd step: at the beginning of the dataflash file look for FF FF FF FF, and replace them with 64 00 00 00.

slavyk, 2016-08-12:
Thanks, TBXin! Now it's 1.00!

ClockSelect, 2016-08-17:
Sorry for the inconvenience -- loosing the Hardware version is the worst thing that can happen on that box. This should not happen any more (hopefully).
@TXBin: thanks for the recovery method.
Question is: was really the HW version 1.0 before? Some version differ only by the atomizer shunt resistance value, and reseting to 1.0 of it was, say, a 1.01, will set the shunt resistance value off by .1 ohm, with no other visible effect. All semms to work perfectly. But it will shift all power/temp values by an unpredictable amount.
I guess it's really important to know what is the HW version of the box before messing with the firmware in any ways.

slavyk, 2016-08-17:
Yes, it was for sure HW 1.0 before.

slavyk, 2016-08-17:
I try the latest version... 
[image]
After flashing: 
[image]

TBXin, 2016-08-17:
> I try the latest version... 

@slavyk can you try to upload firmware in the safe mode?
1. Remove battery
2. Press and hold plus button
3. Connect to the USB
4. Release plus button
5. Device will be recognized with 0.00 firmware version

Then upload firmware from file.

slavyk, 2016-08-17:
Hi! It was not appropriate to use safe-mode. I uploaded the previous version without any problem. All works fine now.

ClockSelect, 2016-08-18:
What do you mean by "previous version"?
Is there anything preventing you from using the latest version of the software?

TBXin, 2016-08-18:
> NFE it's not suported anymore

@slavyk  Что ты имеешь ввиду? О_о

slavyk, 2016-08-18:
I failed to write with NFE-Firmware Updater the new versions of ClockSelect.

TBXin, 2016-08-18:
@slavyk it's very strange. Just now successfully uploaded myevic.bin and myevic_enc.bin (latest version from repository).

slavyk, 2016-08-18:
Yes you are right! It is very strange! I tried again, right now. It was uploaded without any problems. In the morning I tried several times, but it appeared to me the same error message as in the pictures above.

TBXin, 2016-08-18:
@slavyk sometimes USB interfaces stucks, and this is OS issue i guess. After a few minutes all starts working fine.