ClockSelect/myevic #3: [Suggestion] Menu improvement

TBXin, 2016-08-18:
At current moment each menu and submenu contains item "Exit" that closes entire menu at all.
It would be better if each submenu item will contain "Back" item, that will returns to the previous menu.
So we get convenient navigation.

ClockSelect, 2016-08-18:
I though about it a while ago.
Most of the time, you enter menus to setup one item and exit. It's very rare you have to do several things in a row. The back option would be really usefull only when you enter the wrong menu, or when you explore all the menus to see what's inside.
Replacing Exit by Back would make things unconvenient since it's a long way to go back vaping, and adding Back to Exit would make menus a bit heavy, imo. I'm not sure the benefits would be worth the added complexity.

TBXin, 2016-08-18:
@ClockSelect Yes, you are right in part, replacing "Exit" with "Back" will be annoying in some cases. How about to add additional item "Back" before "Exit"?

ClockSelect, 2016-08-19:
I considered the case, if you read carefully^^
It would make menus cumbersome. They are not enough deeply nested to "Back" beeing really be usefull. It would add one click to go to the Exit option in most case. I'm perfectly aware of your concerns; I had this very same thinks when testing my menus.  I don't think we should do this.

closed 2016-08-23 by ClockSelect