ClockSelect/myevic #5: Compile

Daveee10, 2016-08-28:
Hello,
I've got some problem. I don't know which packages and compillers needed. I think not a bad idea how you make a list with required packages and tools with versions what U are using for compiling. I've got some problem with python-evic and evic-sdk because I don't find a good instructon for ubuntu/mint.

Now I get this error when I try to compile:
evic-sdk/nuvoton-sdk/Library/StdDriver/src/usbd.o: In function `USBD_SwReset':
usbd.c:(.text.USBD_SwReset+0x22): undefined reference to `memset'
make: *** [myevic.bin] Error 1

Thank for your help

ClockSelect, 2016-08-29:
Yes, I've been a little lazy about needed packages.
In fact they are exactly the same than for [ReservedField's evic-sdk](https://github.com/ReservedField/evic-sdk)
And installation instruction for evic-usb should be found on the [Ban3's page](https://github.com/Ban3/python-evic)
I followed the instructions on those two pages and it works fine.
I do not get your memset issue, so I cannot really gues what you're missing...

Daveee10, 2016-08-29:
Ok. I made it too. I think the problem is the gcccompiler because i get c99 issue what fixed with cflag. Imho older version of arm-none-eabi gcc working good bit yesterday not tested

eugenesan, 2016-08-29:
I've posted https://github.com/ClockSelect/myevic/pull/6 with the fix

ClockSelect, 2016-08-29:
Great job!
I did the merge.
Tell me if it solves the issue; I don't have ubuntu to test.

Daveee10, 2016-08-29:
Issue solved. Thx eugenesan and ClockSelect!

ClockSelect, 2016-08-29:
Welcome. All credits to eugenesan.

closed 2016-08-29 by ClockSelect