ClockSelect/myevic #7: Temperature mode seems to be working incorrectly

eugenesan, 2016-08-29:
Temperature mode seems to be working incorrectly.
Setup:
	Device name: eVic-VTC Mini
	Firmware version: 3.03
	Hardware version: 1.00
	TEMP-SS316@0.52Ohm/20W

original3.0.x:
	Provides stable vapor (with PERIODIC "popping" sounds) already at ~195C.
	Average actual wattage is ~19W.

myevic:
	Starts to provide "useful" vapor (with RANDOM "popping" sounds) only at ~250C and is unstable.
	Protection triggered almost immediately after pressing fire button.
	Actual wattage is <5W with random jumps to maximum.
	It seems like resistance is climbing much slower than on original firmware.
	Power mode, with same conditions, is working as on the original firmware.

ClockSelect, 2016-08-29:
Yes; SS316 TCR value has been lowered from 120 to 88, which is closer to the actual value for the alloy. I experimented that it gives far more accurate temps in a majority of cases, but TCR of 316L can vary widely from build to build, even using the same wire. It's a weird thing. It's often better to use TEMP TCR mode with SS316L.

eugenesan, 2016-08-29:
I see.
Thank you for the explanation.

Now, since we are talking about TCR configuration, I wonder if it's possible to move TCR setup screen to "Menus"?

P.S.
I wonder if you could also give me a hint where to look for the corresponding values in order to do the following:
* Sets the temperature adjustment step to 1 degree
* Set menu blink frequency to 2 Hz

ClockSelect, 2016-08-29:
Blink: In timers.c, there is a (EditModeTimer % 50) condition you can replace by a % 25 if you wish.

TCR Menu: I already thought about it, I'll do that soon, I think.

eugenesan, 2016-08-29:
Thanks!

I've committed both "patches" (links are above).
If you think those changes are worthy, please let me know, I''l post a PR.

Daveee10, 2016-08-29:
Today I will make a test build for Wismec Presa 75W and if it will work tomorrow I'll make a pull request or fix the misstakes...

ClockSelect, 2016-08-29:
I do not have a Presa 75, so please be sure of what you did - test thoroughly. I will not take your 80W modif on mini, though.

ClockSelect, 2016-08-29:
@eugenesan: I'll try this; if I feel it improves the user experience, I'll take it.

We are out of subject of the original issue now; I'll close it. Open new issues for your wishes.

closed 2016-08-29 by ClockSelect