ClockSelect/myevic #9: FiveClicks event (Turn Off/On) is unreliable

eugenesan, 2016-08-30:
FiveClicks event is not working reliably.

I am able to turn my mod off/on only in 1 of 5 attempts.
Some times, after several attempts, my mod end-up with random numbers spread across the screen and require a battery pull-up to get it working again.

Daveee10, 2016-08-30:
That is debug mode. Debug mode on/off 4 click. Power on/off isbworks good but click little slower. Debug mode works in power off and on mode too.

ClockSelect, 2016-08-31:
off and on still work fine, but require *exactly* 5 clicks. Most of the time, when switching your device on and off, you do not really count how many clicks you give; you click 'till it switches. On this version you have to count 5 clicks. 
This way gives easier acces to features requring more than five clicks, but they're not really used atm. Maybe debug mode should be set to higher clicks.
This question is open to discussion.

Daveee10, 2016-08-31:
Why don't move debug mode to menu seems like usb mode?

eugenesan, 2016-08-31:
Thanks for your replies.

I am happy to know what was the cause of "random numbers".
You really don't want 2.5maH device doing "random" things in close proximity to your body :-)

Second, after flashing original (actually patched 3.0.1) firmware and then freshly built myevic, I was unable to reproduce the issue.
If the issue will return, I'll try to provide more information about the events preceding the issue.

Regarding relocation of debug trigger.
I think ability to enable debug quickly and on any screen is very useful.
Moving it to sub-menu will probably make it much less useful.
On the other hand it seems to be a good idea to keep 4 and 6 clicks free to avoid conflicts with such an essential function as FiveClicks. 
Maybe relocate debug to SevenClicks?

ClockSelect, 2016-08-31:
Maybe I could add an Expert menu item to enable/disable the 4-Clicks so that it con no longer be accidentaly activated. Would take a bit in the dataflash flags, but we have plenty available.
I think that adding the switch on/off function to the 6-Click will solve most of the issues; problem on switch unreliability comes essentially from the fact that people click a 6th time.

ClockSelect, 2016-08-31:
Give a try to this version.
5, 6 and 7 clicks switches the box on/off.
Should work with Parkinsonian users. ^^

ClockSelect, 2016-08-31:
The story behind: the OFW switches the box on/off *immediatly* after receiving the fifth click event. That's the reason why you're forced to switch off the box prior to using features requiring 10 or 20 clicks (board temp & fw version), since the 5th click will switch the box back on.
I'd like to keep open the possibility of addind features beyond the 5th click without having to switch off the box, which I find cumbersome. Let's see if this method works.

eugenesan, 2016-08-31:
After switching FourClicks (Debug) off by default and assigning 5-7 Clicks to Power On/Off everything seems to be working fine.
Though, I still believe assigning Power On/Off to 5, Debug to 7 and ignoring 4/6 would provide experience more consistent with stock firmware while still allowing additional functionality at 7+ clicks.

Anyways,I see the issue as solved.
Thanks!

ClockSelect, 2016-08-31:
Ignoring the 6th click is what cause the discrepancy with stock firmware, so switch on/off must be at least affected to 6th click too to be on par with the stock firmware regarding the box behavior.

closed 2016-08-31 by ClockSelect