ClockSelect/myevic #10: Screen mirroring/rotation causes BSOD

eugenesan, 2016-08-31:
Triggering Screen mirroring/rotation causes BSOD (BlackScreenOfDeath).
To reproduce (_steps 4-7 are optional_):
1. Turn off the device (FiveClick)
2. Press and hold + and - buttons until screen rotates
3. Wait until screen goes off
_4. Turn the device on (FiveClick)
5. Do anything
6. Turn off the device (FiveClick)
7. Press and hold + and - buttons until screen rotates_
8. Observe BSOD

To restore normal operation, re-insert the battery.

ClockSelect, 2016-08-31:
It's not due to screen flipping.
Just switch off the device and click on + or - only is enough.
Unfortunatly I don't get to the BSOD, it only freezes the device.
Investigating.

ClockSelect, 2016-08-31:
Fixed.
Was due to incorrect clearing of GPIO D interrupt flags.

closed 2016-08-31 by ClockSelect