ClockSelect/myevic #12: Feature request: Displaying third line on screen saver and charge screens

eugenesan, 2016-08-31:
It would be useful to see third line also on screen saver and charge screens in order to inspect Battery Voltage or Board Temperature.

ClockSelect, 2016-08-31:
Just click fire button... no?

eugenesan, 2016-08-31:
Clicking Fire button will work but seeing the numbers passively would be nice.
The screen is mostly blank anyway and is constantly updated with Clock/Logo/3DCube/BatteryLevel so adding that feature should not affect battery life while providing useful information.

ClockSelect, 2016-08-31:
Keeping it mostly blank is precisely the purpose of a screen saver^^
And we would loose some aestethics.

Daveee10, 2016-08-31:
IMHO the battery voltage and temp is a good idea on charging screen top of the battery indicator...
Another thing. The logo not working. Is this just for me or this is a real issue?

ClockSelect, 2016-08-31:
If you're charching, you should have the charging screen - blank with batt at the bottom.

Logo should work; I have it fine. Tell me more.

ClockSelect, 2016-09-01:
When charging, battery voltage is always 4.2V, so... It's a bit useless.
Why would you want the temperature?

eugenesan, 2016-09-01:
I just checked, BatteryVoltage in 3rd line is showing actual voltage of the battery and not constant 4.2 volt.
Regarding the temperature, since Lithium cells are better charged when cold, it's useful to know how hot the device is to minimize battery wear.

Daveee10, 2016-09-01:
I'm trying with NFirmwareEditor and Joyetech's original firmware updater, but logo not works

ClockSelect, 2016-09-02:
Give a try to the new charge screen. Made it simple.

ClockSelect, 2016-09-02:
A specific issue has been opend for Logo problem.

eugenesan, 2016-09-02:
Great, the issue is solved.
Thanks a lot!

closed 2016-09-02 by eugenesan