ClockSelect/myevic #15: LOGO does not show up

ClockSelect, 2016-09-02:
LOGO is sometimes not displayed.
Under investigation.

Daveee10, 2016-09-02:
It solved when flashing original firmware, reset dataflash, falsh myevic firmware...

Daveee10, 2016-09-03:
Hmmm.... Logo turned off, some firing, power off/on, logo turned on and nothing... Logo not working...

ClockSelect, 2016-09-04:
Yes, Logo works a few time then ceases working. Still not found why.

Daveee10, 2016-09-04:
Ok, I make some test on VTC mini and Presa... Everytime when resetting dataflash the logo works... I'm tried power off, firing, power off, change some settings, flip display and everything and the logo works good... BUT after turn off the logo and back to on the logo never works... I think the problem is in the setting method... I check the code but don't find the problem...

ClockSelect, 2016-09-06:
Idem.

ClockSelect, 2016-09-07:
Fixed.

When you extrat a bit from a bit field, the integer value is not 0 or 1 but 0 or ~0, which is quite different.

closed 2016-09-07 by ClockSelect