ClockSelect/myevic #16: Wismec Presa BSOD

Daveee10, 2016-09-02:
Presa little buggy :)
When uploafing this firmware the device not reboot. In NFirmwareEditor shows: Software version: 0.0 and Boot mode: LDROM....
After removing the battery and place back the issue not solved but when I connect the USB and then put in the battery the device is working fine.. If powering off the device, I can power it on... But if the battery removed and put in the device not booting only when the USB connected witc charger....

ClockSelect, 2016-09-02:
Strange.
Does you version work on Presa?
I did nothing really different.

Daveee10, 2016-09-02:
No-no... My version is same as yours... The issue is same.. I'm very sad... 
The interesting thing, how I tried to patch the original Evic VTC firmware in Hex editor for presa and that's works very good

ClockSelect, 2016-09-03:
I can't go forward with this issue without having the box by hand, I'm aifraid.
This will remain experimental atm.

Daveee10, 2016-09-03:
Write the things for needed you and how can I read/check/test it.

ClockSelect, 2016-09-04:
I had the same issue with a VTwo Mini; box seems to have an issue with the X32 or the RTC.
Give it a try with forcing the gFlags.has_x32 to zero.

ClockSelect, 2016-09-04:
I wrote a run-time test for the RTC at startup to disable X32 usage if necessary.
Give a try.

Daveee10, 2016-09-04:
Good work bro... Now Presa booting everytime when put the battery or upgrade the firmware

ClockSelect, 2016-09-04:
Fine^^ So now you should test every other features, so we can affirm Presa compatibility for good.

Daveee10, 2016-09-04:
Yes. First look it's good everything. I need to test lot of thigs. I used the firmware couple of days. The logo menu is useless on presa or need a program for uploading logo because only some user using evic sdk and linux. I will talk with txbin to make support presa logo in his app..

ClockSelect, 2016-09-04:
use ```evic-usb upload-logo``` command to upload a logo. should work if you add the logo dimensions in the device.py file:
```'W007': DeviceInfo("Presa TC75W", ['E052'], (64,40)),```
around line 65

Daveee10, 2016-09-04:
Yesbi know that but i wrote how what is the problem. Imho lot of users don't use linux and evic sdk so they needed support this function.

ClockSelect, 2016-09-04:
Yes, TXBin should be the guy for that.

ClockSelect, 2016-09-05:
I close this issue.
Please open anorther for specific issues on Prisma.

closed 2016-09-05 by ClockSelect