ClockSelect/myevic #17: [Idea]Preheat

Daveee10, 2016-09-02:
IMO the preheat is a good thing but the power setting is not...
If I replace my tank to another with another resistance what needed another power very annoying thing how I need edit the preheat value...
I think better way how you make some preprogrammed value like this:
Soft : Power + 1W
Normal: Power + 3W
Hard: Power + 5W
Extra: Power + 7W
This values only ideas...

Another idea:
I think a better way for editing the resistances a submenu seems like TCR Set or Preheat...

npocmogu, 2016-09-03:
I would like the ability to adjust power to 1 watt (not 0.1) and reduce the number of taps of the fire button to turn on/off the device and navigate menu (3 clicks on/off, 2 click navigate menu - will be perfect).

ClockSelect, 2016-09-04:
Since 3.03, puff/time counter are reset by a long click on + and - buttons while editing the third line with counter blinking. It's standard behavior.

ClockSelect, 2016-09-04:
I think battery meter is really enough; % won't provide more informations. It's fairly inaccurate anyway, the discharge curve depends on the battery brand and model. If I tell you 50% for 3.6 volts, it can really be anything from 25% to 75%. It would more or less meaningless.

npocmogu, 2016-09-04:
What are you talking about? Sorry, maybe I badly explained what I want? English is not my native language. I was talking about the power adjustment in vw mode. So when you press the + power was increased 1 watt, but not at 0.1 watt. And the second thing that I said to make the evic turn on/turn off after three clicks fire and 2 clicks to change mod (vw/tc/bypass).

ClockSelect, 2016-09-04:
This is not standard behavior, and I try to stick with Joyetech's settings the users are used to.
The best we could do is at some point give the possibility to configure those in a menu, but not set it as the normal behavior of the firmware.

ClockSelect, 2016-09-04:
Resistance editing has been moved in a submenu in lastest release; give it a try.

npocmogu, 2016-09-04:
Ok, will wait.
When will approximately next update?

Daveee10, 2016-09-04:
Ok, now looks better... I think this is a good way...
IMHO the TCR Set menu looks better if it looks like this menu, and the Power + Plus button method useless

ClockSelect, 2016-09-04:
The TCR Screen is already written, so I keep it.
The Power+ button serves no other purposes, so I keep it too.

ClockSelect, 2016-09-04:
@npocmogu: next update is already there.

Daveee10, 2016-09-04:
I understand that but it not looks as same as other settings page.

Daveee10, 2016-09-04:
You can replace the TCR set menu on Power + Plus button with Board temp what works with 10 or 20 click in original firmware, I don't remember...

ClockSelect, 2016-09-06:
I really want my fw to be as close as possible in its behavior to the original.
If I modify things as you say, an issue will soon be opened telling that "TCR Set menu is no longer available".
I known that guenuine fw is not the most user-friendly over the place, but people are used to it. I do not want to scramble habits.

closed 2016-09-09 by ClockSelect