ClockSelect/myevic #24: Battery percentage

jerryang76, 2016-09-07:
This is a very nice firmware, so many great features and I just saw that joytech also implemented preheat in their newest firmware v5.04(Sep 2nd). Did they steal that function code from you?

There is just one Battery percentage feature missing that I used in NFirmwareEditor. 
Is it possible to add?
[image]

I have also attached hex replace code from NFirmwareEditor using Evic VTC Mini 3.01 version firmware.
[3.01_batt_prc7.zip](https://github.com/ClockSelect/myevic/files/458156/3.01_batt_prc7.zip)

ClockSelect, 2016-09-07:
Preheat comes from 5.04. It was a feature a thought to implement a some point, but they did it fisrt^^

About bettery percentage: if you display numbers, they sould really tell something. Problem with bettery percentage is that it's really inaccurate. It strongly depends on the discharge curve of your very brand and model, and discharge curve are usually very flat around 3.6V, so 50% really means anything between 25% and 75%. That's why I find this feature almost useless. A large bar is as informative as a small bar plus an unreliable percentage.

TBXin, 2016-09-07:
@ClockSelect how about to store discarge table in the dataflash? Also do not use hardcoded values for 0% 10% 20% ... 100% percents, instead using configurable pairs, like DNA does (% - volts).
In this case users be able to setup discarge curve for own battaries.

This is how looks my Samsung 25R setup for DNA
````
"Battery Charge (%)","Cell Voltage (V)"
0,3.02000427246094
1.1773669719696,3.10000610351563
4.56118488311768,3.27000427246094
8.50847816467285,3.41000366210938
24.870023727417,3.57000732421875
38.9835968017578,3.64999389648438
70.0488891601563,3.91000366210938
83.6078872680664,4.05000305175781
93.2166137695313,4.08999633789063
97.1728134155273,4.16569519042969
99.9990844726563,4.19476318359375
````

Also i have an idea to create Escribe software analog for myEvic firmwares. Since sources are open and highly customizable will be easy to move any settings to the dataflash.

ClockSelect, 2016-09-08:
Yes, I thought about it.
I wanted to make a menu were you could tell your model "HG2", "30Q", so on, with discharge curves in rom (or even dataflash - why not). But in fact, it's pretty useless; I learned a lot about batteries in the meantime, and the discharge curve is highly dependant of the aging of the battery and many other factors.

The data you're posting will not be the same after 30 more cycles of your battery.
It also depends on temperature, relief time since last high-power drain, and, and, and.

The curve is so flat between 3.4 and 3.8V that the thiniest perturbation will defeat any attempt to tell anything relevant between "Almost full" and "Almost empty".

The default curve of the VTC in not-so-bad, and the battery management of this box is in fact pretty good. You won't pull more than 2~300mAh more from your battery by tweaking the numbers.

It's not worth the pain.

ClockSelect, 2016-09-08:
I'll even tell you more: it was my first purpose when I decided to disassemble that firmware.
I changed my mind. They made things rather smart.

TBXin, 2016-09-08:
@ClockSelect 300 mAh is good achievement, in case when your battery is 2500 mAh maximum.
Yeah, Joyetech creates cool firmwares with pretty smart functionality and capabilities, but my suggestion is more about the correct calculation battery %.
I don't know how DNA it does, but it is perfect.

ClockSelect, 2016-09-08:
I may try to replace the % table of the box by yours and see if it improves accuracy, but it will mostly improve acuracy for users of Samsung 25R.
For the table to be usable by the firmware, it should be sampled by steps of 10%; can you do this? It should contain 11 samples from 0% (3.1V) to 100% (4.1V).
The 0% is 3.1V on the standard firmware; this is where a little gain is possible. This value can be lowered, and the cut-off values of the firmware reevaluated accordingly. But in this part of the discharge curve, the benefit will be very limited, except maybe for people vaping at very low powers (less than 10W). For higher powers, the internal resistance of the battery will immediately make the voltage sag to the cut-off limit (2.9V). Even if this cut-off is a bit lower (it may be dangerous to go below 2.8V), if you vape at more than 20W, you won't really see a difference.
I'm ok to experiment, but I'm pessimistic regarding the benefits.

ClockSelect, 2016-09-08:
Maybe I can interpolate precentage juste like voltages, so having a table by 10% is maybe not an absolute necessity. Would just take more ROM and flash space.
Also, we'd need tables for other popular batteries, at least 30Q and HG2.

TBXin, 2016-09-08:
I think we need better algorithm for battery percentage calculation. Like this:
found 2 pairs, one has higher voltage, second lower. Interpolate to get "position" between this pair and use this value to calculate percent using values from first and second pair.

There are some tables for DNA boards

**LG HG2**
````
"Battery Charge (%)","Cell Voltage (V)"
0,3.11999988555908
4.10263633728027,3.25999999046326
8.67051982879639,3.38000011444092
13.2066831588745,3.45000004768372
25.8071327209473,3.55999994277954
37.9070892333984,3.64000010490417
78.7818984985352,4.01000022888184
96.4613037109375,4.1100001335144
98.4844207763672,4.1399998664856
99.4924545288086,4.17000007629395
100,4.19999980926514
````

**Samsung 30Q**
````
"Battery Charge (%)","Cell Voltage (V)"
0,3.11999988555908
14.5492811203003,3.40000009536743
23.3216419219971,3.51999998092651
53.6583671569824,3.80999994277954
68.035774230957,3.91000008583069
75.2262344360352,3.99000000953674
80.8172607421875,4.03000020980835
93.6057510375977,4.07999992370605
96.8011169433594,4.1100001335144
99.2002792358398,4.15999984741211
100,4.19999980926514
````

**Sony VTC4**
````
"Battery Charge (%)","Cell Voltage (V)"
0,3.13999891281128
0.689058065414429,3.21000695228577
1.99128675460815,3.30999803543091
3.71544981002808,3.38999891281128
5.86761808395386,3.42999291419983
14.4611129760742,3.5099949836731
21.9830780029297,3.55999803543091
48.8593826293945,3.66999793052673
65.5303344726563,3.80000305175781
99.4611358642578,4.19000196456909
100.001457214355,4.21000719070435
````

**Sony VTC5**
````
"Battery Charge (%)","Cell Voltage (V)"
0,3.05000305175781
0.349850982427597,3.10000610351563
1.47545850276947,3.19999694824219
4.17083215713501,3.28999304771423
13.5681343078613,3.41999793052673
23.0201950073242,3.5099949836731
44.8326416015625,3.64999389648438
79.3644561767578,3.96000695228577
94.9282608032227,4.11000108718872
99.6801528930664,4.17999315261841
99.9995803833008,4.19999694824219
````

ClockSelect, 2016-09-09:
Precious tables. I'ill take them.
It's a real project, I need time to realease it.
I'll create a branch.

Johnny147, 2016-09-13:
A menu with batteries list seems to me too hazardous and there is too much batteries plus alots of fake or used 100cycles batteries, well I find tables and battery list not reliable in my opinion +too much different batteries on market. 

May-be that's what you are thinking :
Isn't any way to make a percentage by measuring the difference of Voltage between Unload and Loaded ? Then let's say there is 0.3V difference then battery is at half of it capacity ? 
Because that's the only difference between every batteries but is it a reliable way I don't know.

Or make OS record the differences of voltages
-When battery almost full and in load
-When empty 3v in load (When "Weak Battery" kicks in)
Then so create % for next charge and re-record.
But this way non-aware, or users replacing batteries everyday will not understand. (And my be so-on "battery percentage" would be an option for aware users in Expert menu.

Sorry if that's obvious or wrong.

ClockSelect, 2016-09-14:
I agree.
I'll keep the behavior of standard firmware as long as you don't tell her via the expert menu that you have a specific battery. I assume that if you do that, then you know what you're doing.
Your proposals are complicated, standard behavior do the job well for non-aware users.

ClockSelect, 2016-09-24:
@TBXin: I'm trying your tables - they won't like 'em.

TBXin, 2016-09-24:
@ClockSelect whom?

ClockSelect, 2016-09-24:
Users :) That's because they have a less optimistic-looking over the run than the generic curve.
But hey, they'll get used to it.
Btw, where did you find those curves?

Johnny147, 2016-09-25:
My battery become weak when the original power bar is still at half.
I can't select it but it's a random Samsung 20 or 25A 3000mah
I have a long day hardvaping time but the battery indication is not correct.
Battery is around 3.4V when full and 3v when empty ("Weak Battery" appears) 
What is that strange calculation of the original firmware ?

ClockSelect, 2016-09-25:
"Weak Battery" appears when vaping at the selected power would bring or (brings) your battery voltage below 2.9v. You can see it when selecting the real-time battery voltage on the third info line of the main display.

Starting at 3.4v, this is a 0.6v voltage sag.

Internal resistance increases with battery cycles and usage; the more it's used at high Amps, the more it increases. Good guenuine new batteries starts with around 0.02 Ohms of internal rez, average chinese battery are more around 0.1 Ohms, good used batteries are somewhere in-between.
Having a 0.6v voltage sag would mean a 0.06 Ohms internal rez @ 10 Amps, or a 0.03 Ohms @ 20 Amps; those are rather usual figures. 

3.4v is can never be a full battery; it's an almost empty battery - around 10-15%.
If you read 3.4v for a full battery, then your battery is dead.
A full sane battery should read anything between 4.0 and 4.2v.

TBXin, 2016-09-25:
> Btw, where did you find those curves?

http://dna200.de/kurven_akku

> But hey, they'll get used to it.

This is cool!

mkrajinovic, 2016-09-25:
Could you add the LG HE4 (yellow batteries) table in the next build? 
[LG HE4 3.2SoftCutoff.csv.zip](https://github.com/ClockSelect/myevic/files/491519/LG.HE4.3.2SoftCutoff.csv.zip)
Thanks for the awesome work :+1:

TBXin, 2016-09-25:
> Could you add the LG HE4

And this why i suggest to create "Custom" or something like this menu item, which will be pointing to the curve in the dataflash that can be configured via NFE or directly in the device.

mkrajinovic, 2016-09-25:
> And this why i suggest to create "Custom" or something like this menu item, which will be pointing to the curve in the dataflash that can be configured via NFE or directly in the device.
Even better. Great idea, make it configurable via NFE. It would be kinda hard inputting those numbers directly on device.

ClockSelect, 2016-09-25:
Added HE4 data.
I'll think about the custom.

ClockSelect, 2016-09-25:
I tried to lower the cut-off voltages, but unfortunately we cannot really do that.
Even if the brown-out detector and the low-voltage reset system of the MCU are set to 2.2V, it resets when the battery voltage goes just below 2.8V. There's something between the battery and the MCU that "steals" 0.5~0.6V or so. Maybe a diode.
So it seems there's nothing to gain in mAh on the low-voltage side, due to the electronics.

TBXin, 2016-09-25:
No sense to make cut off to lower than 3.2-3.3v.
Who really needs 5-6 extra 5-8 watts puffs?

If i remember correctly stock firmwares completely shuts down when voltage become lower than 3.0v.

mkrajinovic, 2016-09-25:
Ok, something is not right with that HE4 profile I sent, for example if it is selected it shows 53%, when switched to GEN, it is 74%, but if I go back to HE4 once again now it is 65%. I'm not much of a coder yet, but I will take a look when I get home from work.

TBXin, 2016-09-25:
@mkrajinovic will be great if you can also provide battery voltage for same cases.
Battery voltage can change very fast, especially when you are pressing fire.

mkrajinovic, 2016-09-25:
I will see what I can do when I get home.

eVic75, 2016-09-25:
Battery percentage on main screen, 1% accuracy:
[image]
But when screen turns off, 10% accuracy:
[image]

ClockSelect, 2016-09-25:
Yes, battery charging screen is still at 10% resolution. I may change that if you want, but anyway voltage on charge is off by at least +0.5V so it's not really significant.

ClockSelect, 2016-09-25:
53% to 65% on the HE4 is around a 0.15V gap, it's usual on release from a regular vaping power drain.

eVic75, 2016-09-25:
No, no. Don't change it. I mentioned that only for information. I thought that you hadn't noticed it  ;)

closed 2016-09-29 by ClockSelect