ClockSelect/myevic #26: VTWo Mini don't work

npocmogu, 2016-09-09:
With newer version of firmware VTwo Mini (not vtc mini) don't work. After update black screen and nothing more.

Johnny147, 2016-09-09:
I have injected the new firmware and is working right on mine. Hardware version 1 and I had the same firmware.
Maybe try with the latest Joyetech updater ? And unlocking box after ?

ClockSelect, 2016-09-11:
That's weird. New version *sloved* freezing problems.
Unfortunately, I don't have a VTwo anymore to test thing. And anyway, the VTwo I had worked fine.

ClockSelect, 2016-09-11:
I'm afraid someone with a freezing VTwo will have to investigate on this by himself...

maelstrom2001, 2016-09-12:
@ClockSelect Check line 226 in myrtc.c
It should be `if (!--rtccnt)`
We tried it with npocmogu yesterday and it works for VTwo Mini :)

ClockSelect, 2016-09-12:
You're right.
Well done!

ClockSelect, 2016-09-12:
Anyway, it means your VTwo has an issue with the RTC.
As I've seen on various forums (including Joyetech's), it seems to be a fairly common issue. Hardware misconception.

closed 2016-09-12 by ClockSelect