ClockSelect/myevic #27: Digital Clock

Johnny147, 2016-09-09:
Hello,
Could you place a Digital Clock instead of the analog as screensaver ?
Juste like the digital mod on 4.02
This is the patch but I can't apply it to your 3.03 mod :)
https://github.com/TBXin/NFirmwareEditor/blob/master/src/NFirmwareEditor/Patches/Evic%20VTwo%20Mini%204.02/4.02_DClock.patch

I am addicted to your mod myevic and the only missing feature is the digital clock/date and I would be so happy to have it but don't know how to do it myself.
If possible... A big thanks !

Daveee10, 2016-09-09:
Now I'm working on it

Johnny147, 2016-09-09:
Thanks that's so nice ! It gonna be the perfect firmware :D
Regards

Daveee10, 2016-09-11:
Now your request is ready

Johnny147, 2016-09-11:
Thanks again :D !!!

closed 2016-09-11 by Johnny147