ClockSelect/myevic #34: Preheat clause

ploop, 2016-09-11:
I think, need make a percentage of primary power in preheat, because if we change primary power, we need change preheat power. I think it will be more convenient.

Daveee10, 2016-09-11:
I think too or maybe predefined values like some devices (soft, normal, hard, extreme)... IMO this two way is better then actual method, but ClockSelect want firmware to be as close as possible in its behavior to the original. Maybe I will make another preheatr method what changeable with this in expert menu. Maybe...

TBXin, 2016-09-11:
IMHO preheat defined in percentages more convenient and natural for users. Soft, normal, extreme, etc is not about preheat power, it's about how fast preheat-power will be reached.
Best solution is give users choice what kind of preheat they can use. This options should be able to be changed in the preheat sub menu.

ClockSelect, 2016-09-11:
People considering using preheat function are supposed to known what they do and what they want. This is an advanced feature we can not really "simplify". Using percentage would make the max preheat power out of control of the user. Coils have an absolute power range, not a relative power range. If I modify my main power setting, I modify my vape temperature, not the speed at which I want the coil to heat up. Those are two unrelated things.

ploop, 2016-09-12:
In practice it looks differently, preheat power always has to adjust depending on the primary power, and if we change coil - it is necessary to change the setting.

I am not against the current implementation, it must remain, just make the second option for the user.

TBXin, 2016-09-12:
> Coils have an absolute power range, not a relative power range

Yes, that's right, and this is the reason of the problem of absolute values. When users changing atomizer or coils they also changing power, but about preheat they can forget and this can cause some problems.

> If I modify my main power setting, I modify my vape temperature, not the speed at which I want the coil to heat up.

This is not true. All depends on the materials you are using. I am using preheat 110 watts for 1 sec, when my primary power is 95 watts. This is necessary to warm up coils, and if i don't use preheat it will take about 2.5-3 seconds.

ClockSelect, 2016-09-12:
>> If I modify my main power setting, I modify my vape temperature, not the speed at which I want the coil to heat up.

>This is not true. All depends on the materials you are using. I am using preheat 110 watts for 1 sec, when my primary power is 95 watts. This is necessary to warm up coils, and if i don't use preheat it will take about 2.5-3 seconds.

Ok, but if at some time you lower your primary power to, say, 85W to have a cooler vape, you still may want your pre-heat to be 110W to start vapor production within 1 second. If you lower your preheat power proportionaly, your ramp-up will be back to 2 seconds. That's my point.

TBXin, 2016-09-12:
My opinion is that need to give users (experienced) more possibilities for the device configuration.
That's why i suggest create switch for absolute and percentage preheat values.

If look to the YiHi boards - they give user possibility create power curves for each preset. This is more complex for user way but more configurable.

ClockSelect, 2016-09-12:
Made something.
Limited preheat to 200% base pwr to avoid highly predictible problems.
Giv it a try.

ploop, 2016-09-12:
Excellent! This is what we wanted.
Thank you!

closed 2016-09-12 by ClockSelect

TBXin, 2016-09-12:
@ClockSelect for what reason user can choose preheat value lower than 100%? This is not preheat, this is precold :)
Maybe better will be use an addition? I mean Power + 0% = Off, Power + 10%, Power +20% etc

ClockSelect, 2016-10-01:
To cite someone: "My opinion is that need to give users (experienced) more possibilities for the device configuration."