ClockSelect/myevic #36: Wismec RX200S - RX2/3

bluenazgul, 2016-09-14:
Hi ClockSelect,

Would it be possible to create an Custom Firmware (maybe called myRX) like your myevic for the Wismec RX200S - RX2/3?

Maybe it is possible to get the source for the Firmware the Same way you got it for the VTC.

Kind regards
Andre

ClockSelect, 2016-09-14:
It's almost the same code, so technically the source is there... almost.
There's only a few hardware changes between the VTC and the RX200S and the myevic firmware *could* be made compatible with the RX200S; not an obvious task, but doable.
I don't know about the RX2/3 (didn't look at the code yet) but if it's doable for the RX200S, the RX2/3 should not be a big deal either.
Unfortunatly, I do not own any of those boxes, so it would be a bit difficult to test.

bluenazgul, 2016-09-14:
Sounds great, i believe If you offer a closed "Alpha" or "Beta" there will
be many tester, i am Admin of two German FB groups and I now there are a
lot User who allready asked for an RX200S Version

DumanAlti, 2016-09-14:
Here, im the second voluntary ;)

Tomish666, 2016-09-14:
Yesss...here is the third voluntary! Please make it true!😀👍🏻

Lory82, 2016-09-14:
I'm the next :-)
a Custom Firmware like "myevic" for the Rx200S and other models need the user. Plesse :-)

ClockSelect, 2016-09-15:
I'd prefer one of the most enthusiast of you pulling a branch from the current project and trying.
I may give you hints to do this.
For example, for the RX200S:
- Shunt resistance is 110 mOhm
- Display is SS1306
- There is 3 measures of battery voltage:
*    ADC18	BATT VOLTS 1
*   ADC0	BATT VOLTS 1+2
*    ADC4	BATT VOLTS 1+2+3
    Batteries are serial, so the three ADCs mesures cumulatives sum of voltages.
There is no Boost Converter for atomizer; PWM2 is replaced by a second buck converter running in parallel when firing. PWM0 alone is used for atomizer probing.
There is a Boost converter (PWM5) used to boost voltage from USB port to battery for charging. ADC 3 and 13 are used to control battery charge.
There may be some more things but I did not took much time to investigate.
Not obvious, I said.

ClockSelect, 2016-09-15:
Let's make it short: I do not need a tester, but a developer^^
Whit the box by hand.
Sure testers will be welcome and necessary. After^^

chris180389, 2016-09-15:
Here is a devloper with a Box in the Hand! ^^

ClockSelect, 2016-09-15:
Fine!
I encourage you to fork the project, create a branch, and... Let's try! :)

bluenazgul, 2016-09-17:
@chris180389 do you allready have a new branch created yet?

ClockSelect, 2016-09-22:
I had a look at the cuboid, and it's very close in its hardware to the RX200S.
I do not own a cuboid either, so it doesn't advance the schmilblick very much, but I'm pretty sure the Dual will be very similar, and I'm waiting to buy one.
Once I have it in hand, things may be more clear.

maej72, 2016-09-24:
so you will try to mod the vtc dual and the rx2/3?

ClockSelect, 2016-09-26:
Dual, sure. rx2/3 & others rx will depend on my success with the dual.

bluenazgul, 2016-09-28:
@chris180389  @ClockSelect any Progress here?

ClockSelect, 2016-09-29:
Should have my Dual within days.

bluenazgul, 2016-10-04:
Sounds great @ClockSelect sadly i cant test on Cubiod anymore, but still able to test on VTCmini / RX200S and RX2/3

afol2, 2016-10-05:
I have a Cuboid Ready to test, shouldn't be the same as the VTC Dual ? as it have 2 batteries. Also i have another question, it is known that the cuboid have a Hard Lock on the amp @ 25 making it almost useless at less than 0.3 ohms, this is something that can be bypass ?

ClockSelect, 2016-10-06:
@afol2: Unfortunately no; the 25A is a hard limit, as are the 9V. You have to tweak the rez value to get the power you want.

bluenazgul, 2016-10-06:
Would be a dream that out AT's would Detect the used Batteries (but that
wont work).

I remember that there was an unofficial and non public patch für the Cuboid
to change the Ampere Limit.

I only use Sony VTC4 so 30A are safe for the.

afol2, 2016-10-06:
Hmm... that could be like something that could be added in >Expert Mode< some kind of tweak that allow you to change the Ohm value , even tho it is lower. it also can have like some kind of warning to just uses recomended tested batteries 30 amps, Like LG HB6. @ClockSelect

afol2, 2016-10-06:
Or that you have to select in Expert>Batt>HB6 in order to allow the tweak @ClockSelect

Filo301, 2016-10-10:
It's 25A limit on output, not input. Maybe just dc-dc can't handle more.

ClockSelect, 2016-10-12:
Output values are limited to 9V and 25A.
Put together, it's a power of 5625W, which, IMO, should be enough for anyone here.
The very limiting factor is the power limit enforced by the software (75W, 150w, etc depending on the box), which in turn limits the volts/amps accordingly.
The drain on the battery does not depend on the volts/amps sent to the atomizer by the dc-dc; it's only a function of Power (given the dc-dc effciency, say 90%) and battery internal resistance.
At a given power setting:
0.9 * P = ( U * I - r * I² )
With U = idle battery voltage, I = battery drain, r = battery internal resistance.
Solve this equation and you have the battery drain.
I = ( U - sqrt ( U² - 3.6 * r * P ) ) / 2r
If you do the math for a single-cell box, given a 75W limit and an internal resistance of 30mOhm, it gives you a battery drain limit of around 23A at 3.6V, which is reasonable.
If you think your battery can go up to 30A with no issue, you could, theorically, set your power limit up to 90W.

ClockSelect, 2016-10-12:
Oh, by the way: I've been lent a Cuboid a few days, so...
Now we have the Cuboid :)
This opens the way to others multi-cells boxes, but I can't implement a new box without having it by hand. Multi-cell management is complicated and no two boxes do this the same way.
Next to come should be the VTC Dual, and after... It depends on which boxes I can borrow to someone^^

bluenazgul, 2016-10-12:
Stock Limit on RX200S and RX2/3 is set to 50 Amps by Wismec.

So i would say the "custom Firmware" should keep this limit, as there are
no Batteries that have this limit.

bluenazgul, 2016-10-12:
@ClockSelect is there a way to contact you private? Iam Admin of the
biggest German speaking RX Facebook Group, maybe there could be a way to
lend you a Box ;)

ClockSelect, 2016-10-12:
@bluenazgul
Sure! Give me the link to your group and we'll talk via FB.
(But I wouldn't be surprised I'm already a member)

ClockSelect, 2016-10-12:
50A on the output side. Use the formulas, you'll see it's still around 20A on the battery side. Cells are in series, you have three times the voltage.

bluenazgul, 2016-10-12:
@ClockSelect

Here is the Link ;)
https://www.facebook.com/groups/WismecRX200DE/

bluenazgul, 2016-10-18:
@ClockSelect as promised my Parcel is on the way to you ;)

ClockSelect, 2016-10-19:
I will try not to burn them^^

afol2, 2016-10-22:
so, i was wondering if you find a way to overcome the 25amp limit in the cuboid ?

ClockSelect, 2016-10-24:
Overcoming limits is easy; I more or less just have to change a number somewhere. Problem is: how do I know the actual box limits before failure? The only thing I can be sure of is that the limits enforced by the factory firmware are on the safe side, but remember that it's chinese cost-effective conception. Hardware limits must not be *that* far.  
Remember that your 25A do not pass only throught the atomizer; they also go throught a tiny 100 milliOhm shunt resistance, and that's already 62W for her to dissipate; it's huge. I don't know the dissipation limit of this resistance, but again, they may not have taken a 100% margin.
The 3-cells reuleaux have a higher limit (50A, iirc), but they have *two* parallel shunts to measure atomizer current. So it make me think the margins are not so large.
I'd like to avoid issues like "my box is dead", so for know I don't play too much with the numbers.

ClockSelect, 2016-10-24:
Well... Here we are.
Maybe the more adventurous of you would give it a try :)

Filo301, 2016-10-24:
@ClockSelect  What you exactly need to know about rx2/3 to support it?

Sorry for my bad English.

dywersant, 2016-10-24:
Testing now, so far so good. Great piece of software.

dywersant, 2016-10-24:
Btw you can flash rx200s with rx23 firmware. Tried that, everything worked fine. So supporting rx23 shouldnt be much of a problem.

ClockSelect, 2016-10-24:
RX2/3 firmware handle more things than the one of RX200S, so flashing the first in the second may work fine (albeit it's absolutely not recommended), but not the other way.
If you'd try (don't!) to flash the fw of the RX200S in the RX2/3, it couldn't handle correctly the dual-cell configuration, for exemple. It could just fail, or, do bad things.

I must thoroughly examine the fw of the RX2/3 to spot and reproduce all the tiny difference it may have in battery and chipset management. It will take some days.

Filo301, 2016-10-24:
If you need pics of pcb etc. i can make it.

ClockSelect, 2016-10-24:
Yes, if you have good quality pics, I'll take them. It may help sometimes.

Filo301, 2016-10-24:
Interesting.... Almost this same. I don't have more 200s photos.
23: https://imgur.com/a/MIiSW
200s: https://imgur.com/a/c8Hya

ClockSelect, 2016-10-24:
That's what I though, MCU is under the screen. That's an essential part; on the picture with the screen removed, we only see half of the pins of the MCU. There's something on it hiding the 12 top and 12 bottom pins.
Can this part be safely removed?

Filo301, 2016-10-24:
I already assembled it. Please, wait a moment.

ClockSelect, 2016-10-24:
That's it, grats.
Can't see right now if I can retrieve the full size pic from Imgur since I'm on my phone; I'll examine all that when I'm back.
Thank you!

Filo301, 2016-10-24:
I upload it on MEGA, full size photos. https://mega.nz/#F!U5wzDDID!WP5WfFAV4MitiKFACk5QOA
Thank you for awesome software!

ClockSelect, 2016-10-24:
Precious.
Hope they won't be needed.
But they're here.

ClockSelect, 2016-10-25:
Today's release was awaited...
[image]

afol2, 2016-10-25:
RX200S is the same as for the RX2/3 ? @ClockSelect

bluenazgul, 2016-10-25:
@afol2 it is not the same - RX2/3 is still work in progress

afol2, 2016-10-25:
@bluenazgul oh ok.!, thanks

bluenazgul, 2016-10-25:
@ClockSelect Here are some Pictures of the RX2/3 Mainboard - i hope they
could help you

bluenazgul, 2016-10-25:
@ClockSelect has my RX2/3 on his Desk ;)

bluenazgul, 2016-10-25:
@ClockSelect A friend has two RX2/3 disambled right now (for FatDaddy mod
;) ),  he will send me Pictures later today

e0x, 2016-10-25:
the recent commit say support RX200S , anyluck that this can work in the RX200 too ?

thanks

Filo301, 2016-10-25:
@bluenazgul I aleready upload some photos.
@e0x Main problem is different oled, same as istick pico.

e0x, 2016-10-25:
@ClockSelect make sense , thanks anyway.

ClockSelect, 2016-10-26:
Enjoy! :)

Tehnik45, 2016-10-30:
Привет ClockSelect,RX200S, after flashing boxing from the start temperature of 38 degrees, 3-5 puffs and the temperature 53-56 degrees, any ideas why?

Tehnik45, 2016-10-30:
RX200S, and insert the battery box temperature with cold 28 degrees itself rises to 38-40, and that no steaming !? What to do, help!

falconettt, 2016-10-30:
Confirm that the temperature is increasing, it is not normal, need something to do with it. This was not on the original firmware.
RX2/3

[20 more replies]

closed 2017-01-18 by ClockSelect