ClockSelect/myevic #37: Ohm meter to 3 digits after dot

lojpek, 2016-09-15:
Hi ClockSelect

I see in NFirmwareEditor patch that could do 3 digits after dots in ohm meter and I use that on my presa until I see your firmware. It will be very good to see that in myevic.
And I think that 10/15/30 sec dim would be fine option because 1 min is somethimes too much for me.
Anyway great job :-)

regards

TBXin, 2016-09-15:
Hello,

I am not ClockSelect, but i can suggest to you this firmware: http://www.ecigtalk.ru/forum/f16/t101098.html#post7516028 (2nd post in the forum thread)

This firmware is done by my colleague maelstrom2001 (fork of myEvic with some cool features).
You can use this firmware, until ClockSelect will not add these functions to the main repository.

lojpek, 2016-09-15:
Great :-) thanks i will test it :+1: 

[https://cdn.meme.am/instances/500x/65162406.jpg](url)

ClockSelect, 2016-09-15:
@TBXin: I'm ok with that :)

For the dim, there are two things:
- The main screen dims after 30s and for now it's not configurable,
- The screen saver dims after your configured time in the screen protection menu (and can be set off).

If you are talking about the main screen, I may consider this option. For the screen saver, less than one minute or Off is basically the same. The purpose of this screen is to keep the box awake, in fact.

lojpek, 2016-09-15:
@ClockSelect I mean main screen for that option. I don't use screensaver but less time that box is awake is for me good option because when I vape I take couple shots and hide to the pocket. I rarely turn off my presa because I feel if the main button is pressed. My situation is when I put box to pocket and when it's on sometimes my phone or keys change + or - button and it's not comfortable to dry hit later. When screen is off faster pressing those buttons don't do anything.

ClockSelect, 2016-09-15:
I see. I'll do something.

@TBXin: did not see the work of maelstrom2001 yet, there's some pretty good stuff in it.

ClockSelect, 2016-09-15:
We should be fine for the rez

ClockSelect, 2016-09-16:
Modified Screen Protection menu; give a look.

ClockSelect, 2016-09-19:
No news, good news.

closed 2016-09-19 by ClockSelect