ClockSelect/myevic #38: not use inversion at the choice of menu items.

Janishewskiiy, 2016-09-16:
The request not to use inversion at the choice of menu items.
The OLED screen is subject to burning out therefore very small inverted font can't be read.
Especially it is critical at installation of time and date.

Replace, for example, with blinking, or underlining.

ClockSelect, 2016-09-17:
I aggre for date/time menu, it's not very well done.
But for the rest, it's a bit surprising. Display of the VTC is sharp and clean, even inverted, even at full contrast.

TBXin, 2016-09-19:
@ClockSelect by "burning out" he meant that the most active pixels with time are going to be dim.
Like at this photo: 
[image]

ClockSelect, 2016-09-20:
Well... I see that; but what can we do against?

ClockSelect, 2016-09-20:
And I'm not sure it's what he means by "burning out". I think it's more dark pixels being enlightened by surrounding white pixels, making black lines blurry. But even though, it's hard to prevent any malfunctioning hardware.

TBXin, 2016-09-20:
@ClockSelect 
Him screen is already damaged and firmware can't fix this issue, but can make workaround by switching highlight to the underline or something like :)

At photo that i provided earlier you can see darker pixels at place of the default firmware battery indicator. This is the result of constant USB charging. Screen worked for a long time, which has resulted to that the active pixels become darker.

ClockSelect, 2016-09-20:
For menus, the new font should be more readable since letters are thicker; I wonder if it helps?
And I think I'll rewrite time setting with blinks or something.

Janishewskiiy, 2016-09-20:
Here is a video with a new font.
The problem is partially solved the main menu, but in the clock setting numbers can not be read.

It is very difficult for the camera to obtain the same effect that is visible in reality.
https://www.youtube.com/watch?v=n2KQqE2En6c
https://www.youtube.com/watch?v=yamITKJMJ6Y

ClockSelect, 2016-09-21:
I made the date setting blink instead of invert; it's not perfect but it may help. Give it a try.

How can you have a screen so burnt out? I've used my box 16 hours a day for months and it's still mint.
Maybe they changed the hardware for a better screen at some time.

Also, the font is now configurable in the Interface menu; maybe I won't keep this option very long because of ROM space but for the moment it's here.

Janishewskiiy, 2016-09-21:
VTC mini HW: 1,11 
December 2015

ClockSelect, 2016-09-21:
I completely revamped the clock settings menus.
Should be better.

TBXin, 2016-09-21:
@ClockSelect  hm, strange decision to split the date and time on the independent menu items, but using one screen for configuration both of them.

What i should press to save date or time now? Clicking Fire button just switching placeholders from 1 to 3 and again.

ClockSelect, 2016-09-21:
Long fire to save.
Date and time are split because the date is setup once and for all. You modify time far more often, so don't bother with the date in the time interface. Less clicks.

closed 2016-09-22 by ClockSelect