ClockSelect/myevic #40: dual battery plus 100 watts hack?

maej72, 2016-09-18:
hello dev,
im a big fan of your work and i have an awesome idea and i hope it works 😄 

im not sure about the board specs in the (vtwo mini) and the input voltage, but what if it could handle two 18650 batteries (in series or parallel battery config) this will make its battery life so good and leads us to the next point

so i saw this post about hacking the (vtc mini) to make it operate up to 100 watts 😮 
http://www.ecigtalk.ru/blogs/u24585/post-16949.html

so imagine 100 watts plus running on two 18650 batteries :D 
no need to buy a new mod if this works right?

bluenazgul, 2016-09-18:
Two 18650 in series wont work (i allready killed One Mainboard by trying
this).

But two 18650 in Parallel is possible.
I also know there was an non Public Release Patch to get 100W working with
the Joyetech Firmware

Daveee10, 2016-09-18:
With one battery 80W is OK for some puff then only lower power works...
Two battery only usable in paralell mode... BUT better idea how you change your mod to VTC Dual. That works 75/150W...

ClockSelect, 2016-09-19:
The board is absolutely not designed to handle batteries in series. You'll burn it.
In parallel, your two (or more) batteries will act just like a single battery with high capacity. You won't see the difference, really, except you'll have more vape time. And battery usage (bar or %) will remain the same.

Some work will be done once I have a dual by hand; I'm waiting for it. But the dual is the dual, it'll be design for serial.

@Daveee10: 80W is already too much. Even 75W. The first setting of the first firmware @ 60W was set to be at max current of the best batteries at full charge, regarding their datasheet. Doing anything over 60W will stress your battery.

closed 2016-09-19 by ClockSelect