ClockSelect/myevic #41: evic vtc mini 75 w

kae64, 2016-09-20:
Good afternoon.
 After flashing EVIC VTC mini 75w firmware from the folder bin - myevic_enc.bin (3.10, https://github.com/ClockSelect/myevic)- It pierced through a file of official firmware, then there is no formal regime titanium and S316.
And sews through emergency vostanovlenie (+ clip, and then connecting yusb), using officially 1.02 does not restore modes.
 The official firmware 3.0, 3.01 all modes. -  poower - yes/bypass - yes/start - yes/tempni - yes/temp Ti - yes/ ss316 - eys/tcr - yes/
 Starting with firmware 3.02 mode of titanium and 316 missing. -   poower - yes/bypass - yes/start - yes/tempni - yes/temp Ti - no/ ss316 - no/tcr - yes/
 When you reinstall - myevic_enc.bin -   poower - yes/bypass - yes/start - yes/tempni - yes/temp Ti - no/ ss316 - no/tcr - yes/ - the firmware on your flash when you turn on in the settings, they are not displayed when you switch. included exposing yes, does not help in the menu.
Help solve this problem.
 Sincerely, Alexander

kae64, 2016-09-20:
Installed firmware from joyetech (+ clip, and then connecting the USB), using officially joyetech 1.02 mode does not recover all of the modes. The other user in the same way everything turned out. That is, when you return to the firmware of the titanium are not available mode and ss316.

Daveee10, 2016-09-20:
When you see temp ni press the minus button then you can change the ni mode to ti and ss316 with the plus button seems like tcr mX changing

kae64, 2016-09-20:
titanium and steel are not available in full on the official firmware 3.02 or higher.
Prior to the shutdown , these modes are available . If you turn them off in the settings to re-enable them does not work. They are not available in the firmware 3.02 and above

kae64, 2016-09-20:
Daveee10 . 
Your proposal only changes the temperature

kae64, 2016-09-20:
Such errors are seen in many users have installed 3.10

Daveee10, 2016-09-20:
No... You don't know how the officially firmwares work..
1. Three click
2. Plus button for go temp ni mode
3. Minus button for change select sub mode
4. Plus button for change to ti or ss316 mode
5. Press fire for confirm your selection
6. Be happy

kae64, 2016-09-20:
This I know . None of these modes when switching . They disappeared after applying the custom firmware .

kae64, 2016-09-20:
Firmware https://github.com/ClockSelect/myevic/tree/master/bin not working. I wonder who so tried. A lot of mistakes. Who will advise on how to reset the way to the good state.

Daveee10, 2016-09-20:
That's working... I'm using that...

kae64, 2016-09-20:
very bad work in the power jumps resistance mode, after the first shutdown mode again they are not already included. This is bad. Very bad

kae64, 2016-09-20:
And still I restore my fashion work.

kae64, 2016-09-20:
[image]
[image]
[image]
[image]
[image]
temp Ti - no/ ss316 - no

Daveee10, 2016-09-20:
Maybe my fault but in this firmware everything works fine... I don't use official because this is better....
Check my gallery:
http://imgur.com/a/u8yVl

kae64, 2016-09-20:
Please give your firmware on my mail kae64@live.ru

Daveee10, 2016-09-20:
Why don't download from ClockSelect's myevic tree?? bin folder - myevic_enc.bin file
This is the link: https://github.com/ClockSelect/myevic/blob/master/bin/myevic_enc.bin

I use this on my VTC and my build on Presa... This 2 mods working very well with this FW....

kae64, 2016-09-20:
The program is installed on my mod, but not one button is not working
[image]

maelstrom2001, 2016-09-20:
https://youtu.be/ice0nopNUwU?t=3m59s

kae64, 2016-09-20:
maelstrom2001 A huge thanks , flashed 3.03, it works. )))
The issue can be closed

closed 2016-09-20 by ClockSelect