ClockSelect/myevic #44: Issue with the battery in TC mode

vhsw, 2016-09-21:
On fully charged battery after pressing the fire button in TC mode the battery voltage drops below 3.10 volts for a second and my VTCmini reports "weak battery".

ClockSelect, 2016-09-21:
Your battery is dead.

vhsw, 2016-09-22:
Nope. Battery works perfectly on stock firmware.

ClockSelect, 2016-09-22:
That's weird; battery management is unchanged (yet) regarding the stock firmware and there should be no difference. I never experimented your problem (I mostly vape in TC, even on my RTAs).
Could you give me more information about your settings?

vhsw, 2016-09-22:
Pls, close the issue, problem gone after reinstalling firmware in safe boot mode (connecting device with right button pressed).

closed 2016-09-23 by ClockSelect