ClockSelect/myevic #45: to see puff, time and clock on the screen at the same time

falconettt, 2016-09-22:
Hi,

It would be great for me to see puff, time and clock on the screen at the same time . 
And battery as digit is enough. 
Better yet would be to get the statistical number that reflects the speed of liquid intake.

Thanks.

ClockSelect, 2016-09-22:
The way the 3 info lines are done make their configurability tricky. They are interactive items, used to change TC power or to lock resistance, and I can't think of a user-friendly way of making them easily customizable. Things must be kept the most simple they can be.
Your best option atm is to display the puff time or counter (the one you feel the most useful) and use the double fire click to switch between info lines and the clock when needed. But I guess it's already what you're doing.
Maybe I'll come back to this if I have a good idea about how to do things.

ClockSelect, 2016-09-27:
Since we're discussing about vape velocity, I think we can close this issue.

closed 2016-09-27 by ClockSelect