ClockSelect/myevic #47: eVic VTC mini - problem with connection to NFE configuration

Dareksa, 2016-09-23:
eVic VTC mini HV:1.01 after update to latest version can't connect to NFE configuration, hangs on downloading settings.
With myevic_fork firmware everything works properly so it looks that problem is not related to hw.

mirekwrozek, 2016-09-23:
I have the same problem HV:1.00

Dareksa, 2016-09-23:
With eVic VTwo Mini firmware works OK.

mirekwrozek, 2016-09-23:
so we're waiting

ClockSelect, 2016-09-23:
I works well with the binary currently available (on my VTC mini), but I know sometimes NFE as problems downloading settings. Sometimes restarting NFE works, sometimes reflashing the firmware works. I fact I don't really know what works or not^^ But eventually, it works.

Dareksa, 2016-09-23:
I tried restarts NFE, and reflash vtc mini but I have still the same problem.

ClockSelect, 2016-09-23:
Being stuck on the "Downloading settings..." means that NFE waits forever for the download of the dataflash. But downloading dataflash works perfectly well with other softwares, so this can hardly be an issue with the firmware.

eVic75, 2016-09-23:
I confirm: eVic VTC mini HV:1.01 after update to latest version (23092016) can't connect to NFE Edition-Configuration, hangs on downloading settings. Restart NFE, reflash didn't help

ClockSelect, 2016-09-23:
I just reproduced the problem, solved it by restarting NFE this time.

eVic75, 2016-09-23:
I restarted, and restarted, and restarted... and still "Downloading settings"

ClockSelect, 2016-09-23:
Ok. Found something: when I have the problem, if I change the Main screen dimmer timeout (in the Screen Protection menu) from 30 to 15s, it works.
Strange.

eVic75, 2016-09-23:
From 30 to 15s?  In "Screen - Protection - Main" I've got 5 - 10 - 20 - 30 - 60s

ClockSelect, 2016-09-23:
(or anything)

ClockSelect, 2016-09-23:
Should work, now.

Dareksa, 2016-09-23:
Yes, now it's working :)

eVic75, 2016-09-23:
Reflashed, got 15s option, and still "Downloading settings".

ClockSelect, 2016-09-23:
Try reseting the dataflash?

eVic75, 2016-09-23:
Do you mean: connect without battery with pressed right button? Yes, I tried. Then flashed and still "Downloading settings".

ClockSelect, 2016-09-23:
No, I meant reseting parameters. Use the advanced tab of the Firmrware updater of the NFE.

eVic75, 2016-09-23:
You're great :) Reseting dataflash helped. Thank you :)

Dareksa, 2016-09-23:
Are you going to add temperature dominant function to your firmware ?

ClockSelect, 2016-09-23:
It's in the TODO list.

Dareksa, 2016-09-23:
OK, thanks for your help and very good work :)

mirekwrozek, 2016-09-23:
Now it's working :) Good work :D

closed 2016-09-23 by ClockSelect