ClockSelect/myevic #48: TDom - reverse action?

eVic75, 2016-09-23:
When I set TDom to "On" I've got:
[image]
When I set TDom to "Off" I've got:
[image]
Is it reverse action or I don't understand TDom?

ClockSelect, 2016-09-23:
Well, maybe On and Off should be switched. That's semantics^^
I'll check that out.

ClockSelect, 2016-09-24:
In fact, it's the very purpose of TDom - regulating max power instead of temperature. Your max temp should be fixed to the temp you like to vape; the max power in TC mode acts like a preheat (and indeed is). So It's not really reversing; it's another way of making your vape more fluent.

eVic75, 2016-09-24:
"Dominant Temperature", "Priority Power"... Yes, you were right: that's semantics... ;) I thought: temperature is dominating, so it should be bigger on screen :) Best regards.
[image]

closed 2016-09-24 by eVic75