ClockSelect/myevic #49: Over 10 second protection

falconettt, 2016-09-23:
Hi,
"Over 10 second protection" -  is it possible to make it ajustable?
I don't want but do occasionally vape inhaling for more than 5 seconds.
It would be great to control it.

ClockSelect, 2016-09-24:
If you vape "occasionally more than 5 seconds", you should not be concerned by the 10s protection. The 10s protection is made for people sitting with their box in their pocket, pressing buttons with their keys, and I feel it's mandatory.
10s is **huge**. If you need more, it means you need to change your tank or your coil to better fit your wish.
That's my opinion; we can discuss, but I'm not really open to breaking safety.

falconettt, 2016-09-24:
No need to break safety.
I do not need more, i need less.
Let's make it "Over ? second protection". 
Let me to to decide what is "?", from 1 to 10 sec.

ClockSelect, 2016-09-24:
Hm... Why not. 1s is useless, imo. Let's say 3-10.

ClockSelect, 2016-09-25:
Ok, let's say 2-10.

closed 2016-09-25 by ClockSelect