ClockSelect/myevic #51: Long fire protection

falconettt, 2016-09-25:
Yes, it works. Great. Thanks, thanks. THANKS! -))
But IMHO some bug at the end. 
Mod gives "Long fire protection" and display freeze at this phrase, you need to press fire to return to main screen. 
I think, better no freeeze, pause 3 sec and automatic return to main screen.

Sorry for my english.

Daveee10, 2016-09-25:
That's not bug. The 10s protection works seems like this, only the time now editable.
IMO not bad how the "warning" and "error" messages stay on screen while you don't press a button. IMO....

falconettt, 2016-09-25:
ClockSelect,
I insist on my logic. Because my logic is not for protection, but to limit vaping. 
I vape too much, it's not good. 
If you want protection just live it 10 sec.

Daveee10, 2016-09-25:
Maybe for you a better feature is the puff limit. I think what on SMOK mods. You can set a max puff number and when you reaches that you can't vape, just if you reset the counter or extend the limit... I thinking on this feature. Maybe ClockSelect will tell his opinion or make his modificaton on the long fire protection. Maybe :)

falconettt, 2016-09-25:
IMHO puff limit is not convenient. Better some statistical number (some math combination of time and puff). Let's call it "velocity of vaping". If number is low - you vape. If number is high - you do not. And you decide what is high and what is low.  

Other choice:
The speed with which you drain the battery is "velocity of vaping".
JUST SHOW IT ON MAIN SCREEN!

ClockSelect, 2016-09-26:
What do you mean by "vaping too much"?
Sounds like a nonsense to me.

falconettt, 2016-09-26:
I'll explain the SENSE. 
There is a need to control the amount of evaporated liquid. This somehow implemented in all advanced devices. First, the length of inhaling (you almost did). Secondly, how often do you do these inhaling (speed). How it can be implemented? For example, like this: 
http://www.joyetech.com/product/control-head/
[image]

Look at smile icon.

ClockSelect, 2016-09-26:
Yes I think I see what you want.
Maybe it could be implemented as an advancement bar on the third line like:
VVEL [|||||||    ]
Telling you how far you are from your limit.
Option would be available only if you configure a limit.
Quantity of evaporated liquid is a time integral of the applied power, have to made some research to approximate it correctly.
A suitable limit would be in mL/hour, for example? I think the number of puffs has no incidence, really - it doesn't tell you how *much* you vape at all.
That needs discussion.

falconettt, 2016-09-26:
Я вижу мы наконец поняли друг друга. Это радует.-)). 
I see we finally understand each other. This is good.-)).
Yes, time integral of the applied power best represents the speed of consumption.
IMO: there is no need to establish a physical limit and block vaping, just show the speed figure on main screen or even better on screen saver.
I'll test your invention and you will decide later what to do next. Okay?
Yes, (mL/hour) is suitable. May be (ml/day) is also useful.
Yes, the number of puffs doesn't tell you how much you vape at all.

ClockSelect, 2016-09-27:
The main question is how to make it reliable.
Displaying a useful information with 64 pixels and when and where is another issue.
Thinking about it.

ClockSelect, 2016-09-27:
First rough estimation: a full 4ml tank took around 1 minute to empty at 30W.
Didn't truly realize before testing^^

falconettt, 2016-09-27:
Look here:
https://en.wikipedia.org/wiki/Joule

OK, full 4.5ml tank took around 10 minutes to empty at 30W

10 min = 600 sec

600s x 30w = 18000 J

You have vaping time in time counter.
You have day time because you have clock.

Suppose now 16 o'clock. It's 57600 seconds from the beginning of the day.

VVEL = 18000 J / 57600 s = 0,3125

Suppose now 17 o'clock. It's 61200 seconds from the beginning of the day.

VVEL = 18000 J / 61200 s = 0,2941

Suppose now 18 o'clock. It's 64800 seconds from the beginning of the day.

VVEL = 18000 J / 64800 s = 0,2777

You see, VVEL is quite reliable this way.

ClockSelect, 2016-09-27:
Yes but at 1 o'clock VVEL = 7. What would it mean?
Measurement should be made in a sliding time window.

falconettt, 2016-09-27:
Suppose now 1 o'clock. It's 3600 seconds from the beginning of the day:

If you do not vape at all:
VVEL = 0 J / 3600 s = 0

If you vaped full 4,5 ml:
VVEL = 18000 J / 3600 s = 5  (you see 5 and, WOW, you understand that it is time to slow down)

If you make "sliding time window" - even better, but let it be as long as possible.
I want to know my VVEL for the week.

falconettt, 2016-09-28:
May be better this way?
VVEL = 0 J / 60 min = 0
VVEL = 18000 J / 960 min = 18,7
VVEL = 18000 J / 1020 min = 17,6
VVEL = 18000 J / 1080 min = 16,7
VVEL = 18000 J / 60 min = 300
VVEL = 18000 J / 10 min = 1800

We have a range of variation VVEL from 0 to 1800.

What would it mean?
Suppose you want to vape not more than 10 ml a day.
10ml needs 22 min of vaping time and 22x60x30=39600 J of energy.
24 hour = 1440 min.
VVEL=39600 J / 1440 min = 27,5

Just keep your VVEL not more than 27,5 and you get not more than 10 ml of liquid a day!

JUST DO IT!   -))

falconettt, 2016-09-28:
You'd probably have to implement it some other way.
But think of it as the speedometer in the car.
If you want to drive 60 kilometers  in one hour you have to keep an average speed of 60 km/h.
I hope this is clear...

ClockSelect, 2016-09-29:
I made an experimental VVel on the third line.

falconettt, 2016-09-29:
GREAT!
I'm testing...
First impression - that's what I wanted to get!
I do not swap batteries but charge via USB.

Can you give some instructions? 

Can I change atoms and wattage?
How to reset and start from the beginning?
Does VVEL starts again automatically  from the begining at 00:00 o'clock?
Does it need tarification?
And so on  ...

Definitely need to put VVEL on screensaver.
What is the accuracy?
Even if the accuracy is not good and I get 12 instead of 10 milliliters of the desired. The next day I will try to get 8 on the screen and I get 10 in reality.

ClockSelect, 2016-09-29:
Atm, calculation is made from box startup.
One may reset value by reseting the box. There is no automatic reset.
You can change ato & wattage as you want. Should also work in temp control, or any other mode.
I have no idea of the accuracy, it needs more testing; but it shouldn't be completely off.
What do you mean by "tarification" ?

falconettt, 2016-09-29:
tarification:
Full 4.5ml tank took around 10 minutes to empty at 30W  IS NOT the rule.
It's all different on different atoms and in different circumstances.
Additionally, you may experience inaccuracy of changes in the ambient temperature.

Can I reset time and puff or it will spoil the measurement of vvel?

Definitely need:
1. Put it in readme, we need more testers.
1. Manual reset any time you want.
2. Automatic reset every day (if you want) and at what time you want.
3. Put VVEL on screensaver.

OК, let's test as it is for a few days...

falconettt, 2016-09-29:
2 hour test result:
How to reset the box properly?
I've uploaded soft once more (I hope this is  also reset) and start.
I kept vvel = 24 ml/d on the screen for 2 hours. It is 1ml/hour velocity.

2 ml of liquid vaped in 105 min instead of 120 min. Accuracy is 87%. Not bad.

I'll start 24 hour test now, you'll get the result tomorrow.

falconettt, 2016-09-30:
Good news!
 
24 hour test:
Uploaded soft to reset and started.
I didn't change anything during the test. (atom, wattage, air temperature and so on).

[image]

You see, it has accuracy about 80%. And accuracy doesn't change during the period of test!

Just simple input for linear compensation will make VVEL accuracy perfect.
IMHO: Definitely needs an automatic reset each day and manuall reset if necessary.

If you go outside in winter or start to change something dramatically - may be the result'd be worse ... 
But I usually don't.

What's next?

ClockSelect, 2016-10-01:
What was your test config? (coil type, power, etc)
I'd like to know if there's significant differences between setups.
Should be tested in TC mode, too.

If you want to rest your box, there should be a pinhole on the bottom to access the reset button with a needle or an unfolded paperclip (it's the case on the VTwo, don't know for other boxes). Anyway I'll add a reset function.

falconettt, 2016-10-01:
Test config was:

evic vtc mini
harware version 1.01
battery sony vts4 in GEN mode.
usb charge and passthrough.
contrast 0%
Stealth ON.
SS 316L / 0,4 om / 18-19 w
VW without preheat.

falconettt, 2016-10-01:
You need MORE TESTERS.
Can try Kanthal and TC mode, but I can't try 50w cause I'll die at this wattage ... -)

falconettt, 2016-10-01:
TCR 120 / SS 316L / 0,28 om / 20-22 watt / 220-240 C.

[image]

You see, the result is just the same.

I'll try kanthal on monday.

falconettt, 2016-10-04:
Bad news,
Uploaded new soft to reset and started.
I noticed you made some changes...
VW with preheat, kanthal, 2 coils, 0,65 om, 26,5 watt.
[image]

Reseted VVEL and started again.
The result is the same.
[image]

What's next?

ClockSelect, 2016-10-05:
I did not make changes (well, I mean wanted changes), but I noticed something wrong too.
Investigating.

High wattage is not the point; people vaping at 60+ watts do not care about their consumption.

TC is more relevant; I think it's where we should be the more accurate.

falconettt, 2016-10-05:
Stopped testing. Please let me know when you find the solution. What test should I begin?

PaoloW8, 2016-10-05:
ClockSelect, I can tell you that I noticed that VVEL stopped working after this commit
https://github.com/ClockSelect/myevic/commit/30b833169eccfdb3c3a927ef816df8e33386574d

I hope this information can help you in some way

Daveee10, 2016-10-05:
Maybe this line cause the problem in mainview.c:
vv /= 10;

ClockSelect, 2016-10-06:
Right: didn't "move" the vv / 10, duplicated it.

ClockSelect, 2016-10-06:
So I changed a few things, beyond fixing the divide-by-ten issue:
- When you reset the ml/d counter, the reference date is set to today midnight; so best time to reset counter is in the morning before starting vaping. Obviously, the counter will not be accurate before a full day passed. It should remain accurate after that, and will show you your average consumption over the period. It's a slowly-evolving counter.
- Counter is now memorized and should survive resets and battery swaps (but since it uses RTC hardware, remember battery must be changed fast on non-RTC enabled devices).

I'm planing on an option to display just total milliliters vaped since last counter reset, just like puff and time. Will make tests simpler.

falconettt, 2016-10-06:
IMHO:
No matter what is the accuracy by itself. The main thing that the accuracy obviously will change (more or less) during the  period of measurement. The longer the measurement period the greater the error. And accuracy will be different for different mods and for different temperature and so on. This counter will slowly accumulate error. 

- At this time, I don't need the period to be longer than 24 hour. What for? 
- I want to be able to carry out linear correction, just some number input for real velocity.

Understand that your tasks, perhaps much wider...

So, lets test!

ClockSelect, 2016-10-06:
I've added the option to display total mL instead of mL/d; some people may prefer this option. What's more, it will permit to judge the accuracy of liquid measurement.

@falconettt : A Watt is a Watt; there's no reason to have high discrepancies between setups. I'm curious we measure that; can people give us their measurements of mL in diffrent setups? like temperature control? Reset the counter, fill a tank of known capacity, vape the tank and check.
The value set up for consumption is the initial 25nL/J.

For the period: the best to do atm is to reset your counter when waking up in the morning. Not a big deal. For configurable reference time and corrective factor, I don't want to make complicated things, menu etc, for what remains a secondary feature imo. But future will tell.

ClockSelect, 2016-10-06:
I upped consumption ratio to 360mL/kJ, this fits my setup, so it should be a good starting base for tests.

falconettt, 2016-10-06:
PERFECT!
Quick test 1 ml:
VW with preheat, kanthal, 2 coils, 0,65 om, 26,5 watt
1 ml in 62 min.
started 20:10,   ended 21:12
its 1272 min from the beginig of the period.
1/1272*1400=1,13 ml/day
vvel = 1,13 ml/day
accuracy 100%
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!

what's next?

ClockSelect, 2016-10-06:
Your correct reading will be at midnight. But since I upped the ratio by 25% and you read at around 75% of the period, you've just been lucky.
I encourage you to check raw mL ATM, not mL per day. We will handle velocity in a later time.

falconettt, 2016-10-07:
Let's make it this way:

We have two modes: "test mode" in ML and "measure mode" in ML/DAY.
You start in "test mode", reset it to zero and vape 1ml. 
You see on the screen some digit other than 1.
You adjust  with (+/-) buttons  this  digit to be 1ML ( adjusting consumption ratio actually, without even knowing what is it).
You switch to "measure mode" and get your pleasure with accuracy 100%.

Yes?

falconettt, 2016-10-07:
02:00   Uploaded soft with  consumption ratio 360mL/kJ,
11:00   Played with and reseted ML mode.
12:58   Reseted ML mode and started 1 ml (SS 316L / 0,37 om / 20,8 w/ VW with preheat 170%/0,3s.)
13:40   Ended test. 1 ml in reality, 1,03 on the screen.

Accuracy 1,03/1*100%=103%. 
3% difference can be neglected. 
So, great again !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!

falconettt, 2016-10-07:
18:21  reseted, started 1ml Nichrome Ni80/1,09om/16,9-18w/VW with preheat 170%/0,3s.
19:40 ended with 1,09

accuracy 1,09/1*100%=109%

Not bad ..........................

falconettt, 2016-10-08:
SS316L/0,38om/20,9w/VW with preheat 170%/0,2s
2ml, 2,09ml on the screen.

accuracy 2,09/2*100%=105%

It's time to let me correct consumption ratio.

ClockSelect, 2016-10-09:
Now you can adjust the vaping ratio in mL/kJ; not the most practical since you have to do the maths, but eh.

falconettt, 2016-10-10:
I do 24 hour test now and everything is going great for now. I'll be done in 6 hours and tell you the result.

falconettt, 2016-10-10:
24 hour test:
I don't think there is a need to discuss the accuracy. It's on the side of the user and "test mode" now. We've got a completely accurate result.
I can say that the 24 hour result is about 96-104% and it's GREAT!
That means if you want to get 10ml/day you'll get about 9,6-10,4 ml/day.

Is's all about 20-40 C temperature om the board.

What about to configure reference time and automatic reset every day?

ClockSelect, 2016-10-15:
I'll have to do a whole sub-menu just for the vvel^^

Daveee10, 2016-10-15:
Want to make some configurable thing for VVEL or why? Just asking.

Other I don't want create other issue for this idea:
Maybe the Game is a complete menu instead of sub-menu of screen because this is not screen setting.
Another: I will happy if you make some other date format option seems like YYYY-mm-dd because we use that... (I haven't got idea for the simpliest way)
Third: The original VTwo Mini style analog clock on full screen (Maybe you have the code for that from decompiled VTwo (mini) firmware) :)

falconettt, 2016-10-15:
If you don't want - I won't insist (everyone has the right to do what he wants)^^
Maybe it's right to keep things as simple as possible...
Maybe later you'll try to improve it.
I like VVEL as it is, and thanks for your work, by the way. 

If you forget to reset VVEL manually every day it becomes unreliable sooner or later. We have not verified the accuracy over a very long period. We understand this. But some people understand nothing, they will complain.

If you want to keep VVEL _"experimental"_ - tell me what to experiment further...

ClockSelect, 2016-10-16:
There's no reason VVEL would become inaccurate along the days, as long as you keep the clock on time.
But it will give you your average per day for the whole period, not for the current day.

falconettt, 2016-10-16:
OK. What will be with accuracy if I start to change batteries over a long period? 
Suppose old and new ones and from different manufactures?

ClockSelect, 2016-10-19:
Should not change anything.

falconettt, 2016-10-26:
Uploaded _"bc7495"_ in RX2/3 . Still alive ...^^
ML counter works but ML/DAY COUNTER DOESN'T. Just 00:00 ml/d. 
please fix this

Evic mini : both vvel counters work OK.

Daveee10, 2016-10-26:
Set the date and time...

falconettt, 2016-10-26:
already did, doesn't help

Daveee10, 2016-10-26:
That's weird... ml/day calculated from epoch time and ml... So that's need to work...

PaoloW8, 2016-10-26:
you have already tried to manually reset the ml/d counter ?
maybe after the reset will start

ClockSelect, 2016-10-26:
on the rx2/3: I've set up date/time, reset the counter, and the ml and ml/d started to work fine.

falconettt, 2016-10-26:
Yes, started, thanks

Filo301, 2016-10-27:
How to reset vvel? i mean total vaped liquid.

falconettt, 2016-10-27:
3 click fire, 1 click -, fire long click, and you see zero

DeDiee, 2016-11-02:
Hello! Sorry for my english. I began testing VVEL on CUBOID. Resistance 0.17, power 80W. With the installation of 360 ml / kj accuracy is 95-97%, continue to observe.
The only negative that does not allow the test for a long time, it is reset ml when I change the battery. I try to do it fast, but it does not always work. What can be done?

[24 more replies]