ClockSelect/myevic #52: Screen Recovery

NikitaAtamanchuk, 2016-09-26:
Hello! I really like the firmware, thank you for what you are doing.
I have an idea. On eVic installed OLED display, and the pixels on them burn out. This is especially the case in the battery meter zone. I suggest add "Screen Burn" or "Screen Recovery" mode - when device is turned off and charging, all of the pixels on the screen will be switched on and off. I think if you leave eVic work in a couple of nights mode, the brightness of all pixels will be equal :)

ClockSelect, 2016-09-27:
I see. OLED is not plasma, so I don't know if ti'll work, but I may give a try.
I'll made a "Snow" screen saver, you'll tell me.

ClockSelect, 2016-09-27:
Give it a try... tell me.
In my experience, Snow is the best way to recover a burnt-out screen.

ClockSelect, 2016-09-27:
And it's another cool-looking screen saver :)

NikitaAtamanchuk, 2016-09-27:
You're awesome, thanks! The idea is to burn off those pixels that are brighter, and then slightly increase the contrast. A few days later, I'll let you know the results.
[image]

eVic75, 2016-09-27:
"Sno**v**" in Menus > Screen > Saver  was quite intentional or mistake? ;)

NikitaAtamanchuk, 2016-09-27:
I also noticed it, but did not attach importance :)

ClockSelect, 2016-09-27:
It's been corrected :)

ClockSelect, 2016-09-27:
Made an Invert option in the Screen menu, just in case...
May help.

NikitaAtamanchuk, 2016-09-27:
You are a genius! :)

eVic75, 2016-09-27:
Battery charging screen is removed?

NikitaAtamanchuk, 2016-09-27:
Not yet. I think, need more time. My eVic used for 6 months.

ClockSelect, 2016-09-27:
@eVic75: no, but it has been included in the "stealthed" screens.
You may set Stealth Off to see it again, or press fire to see batt charge.

eVic75, 2016-09-27:
Ooo... I see :) Thanks for explanation.

NikitaAtamanchuk, 2016-10-06:
Hello! Your innovation produce results! In the combined use "Invert" on the maximum contrast and "Snow" screensaver, the display is almost cleared from the burned pixels. However, in the battery indication area they still a little visible.

ClockSelect, 2016-10-06:
That's awesome!
I must admit I was a little pessimistic on the subject; it's really cool if it works.
For the battery thing, don't use the battery percentage option. Since you burnt the pixels without the option, invert without also. All things must be kept equal^^
I'm happy it works. Good thing.

NikitaAtamanchuk, 2016-10-06:
Yes, I understand :) I burned pixels in that zone when charging device every night (before I found your firmware).
I am also very happy, thank you. A little later, I make a photo of the screen.
I have a request about the battery voltage. Could you please return the battery voltage in the third line? For example, I use a conventional battery indicator, without the "v" and "%" with the battery voltage in third line.

ClockSelect, 2016-10-06:
Things evolves.
Since I made the battery voltage available next to the battery, it becomes redundant on the third line.
What's more, there is already too much options on this famous third line; 8 different things is enough, IMO.
Users should not click 10 times to get what they want; this third line is a problem.

ClockSelect, 2016-10-12:
I agree battery voltage on third line is useful. I restored it.
Will handle the third line problem later.

ClockSelect, 2016-10-12:
I think I can close this issue know :)

closed 2016-10-12 by ClockSelect

NikitaAtamanchuk, 2016-10-12:
I'm sorry, I forgot to upload the photo :)
[image]

ClockSelect, 2016-10-12:
That's awesome!
Still need some time for the battery part, but the result is far beyond my expectations.