ClockSelect/myevic #55: RTC on 3rd info line

marvinerzz, 2016-09-29:
the Real time clock on the 3rd info line gets stuck on displaying the time of last activity(like if you press a button at 00:00:00 it stays there until you press another button, and so on).

ClockSelect, 2016-09-29:
I dot not have this problem; or maybe I didn't understand what you mean.
What I recently changed is about the PUFF counters (count & time), now you may reset them with a long fire click, and after that, they stay in editing mode (blinking) so that you still can change the 3rd line item. It make it easier to reset both counters at once.
But nothing changed regarding the RTC.

maelstrom2001, 2016-09-29:
The main screen must be refreshed periodically while displaying clock in 3rd line.
Adding `ScreenRefreshTimer = 5;` after drawing time procedure (in `case 7: // Real-time clock` of mainview.c) eliminates these "freezes".

ClockSelect, 2016-09-29:
Wrong way^^
Clock moved from 6 to 7 some time ago and the function IsClockOnScreen() in screens.c has not been updated. The dfAPT == 6 should be a 7.
Will be fixed in next release.

ClockSelect, 2016-09-29:
Should work.

closed 2016-10-01 by ClockSelect