ClockSelect/myevic #59: S-mode

falconettt, 2016-10-03:
ClockSelect,
Just for fun and different experience of vaping.
Shall we try it?
Look here:
https://en.wikipedia.org/wiki/Modulation

If simple, something like this:

[image]

But there are more complex variants.
May be good idea to have A and T adjustable.

Also possible benefit to clean the coil.

Daveee10, 2016-10-03:
First look this like a low frequency PWM... Some mode use 7Hz PWM modulation. Maybe 2-3Hz make feelable hot-cold points when vaping. Maybe... Or YiHi TC mode seems like this.. I think not a big work to programming this method.. But ClockSelect knows this better, because I'm not a big C programmer

falconettt, 2016-10-03:
Yes, I mean low frequency modulation and hot-cold points should be feelable.

ClockSelect, 2016-10-04:
Well... If someone wants to pull a branch and try for himself, why not... (anyway, anyone do what he wants). But I don't really feel enthusiast abut this project. What's the point? What would it bring exactly to vaping experience?

falconettt, 2016-10-04:
I'm just asking for the opportunity to try...
What thermal control brings exactly to vaping experience? IMO fun for beginners. I'm happy without TC.
But you can use TC if you want.

You can't say exactly before you try.

But if you don't want to try - let's close the topic.

ClockSelect, 2016-10-04:
Power curves have been on the market for long - Since the very first eVics.
Gonna look a review of this thing.

falconettt, 2016-10-06:
I will try to convince you...
Нот and cold points have different taste.
Something tells me (what is it?) that this algorithm 

[image]
can provide well, just a "rainbow" of tastes.
Isn't it amazing?

ClockSelect, 2016-10-08:
That's more or less the curve of the actual VTC firmware.
One of my projects is to correct this situation, not to amplify it.

falconettt, 2016-11-13:
So what? You did it great man? Or have I just found? However ready.
Here we are: Overregulated PID as low friquency PWM!

My mod now works like a locomotive and is clearly audible!!

Please forgive my rude behavior. 
I have a triple overdose of nicotine as a result of the tests and that makes me "gav-gav"...

With such a toy I do not insist on power curves any more!

P-5000/I-850/D-20
[image]

ClockSelect, 2016-11-13:
Indeed, you can create steady oscillations with the PID.
Didn't tought of that.

ClockSelect, 2016-11-14:
Discussion moved to #144

closed 2016-11-14 by ClockSelect