ClockSelect/myevic #60: Wismec Presa TC 100 W supported?

dabugen, 2016-10-03:
Hi Guys,

big thanks for that amazing work on that firmware, it´s brilliant already and so cool to be able to update to a new version almost daily - so nerdy :) I love it!!

I am about to purchase a Wismec Presa TC 100 W and was wondering if the firmware would also work on that device? I don´t really need the 100 W, just because of the size as 25m clearomizer looks crap on the Presa TC 75 W which I already own. So I am wondering, if I do not need the 100 W capabilit,y, can the My Evic firmware also be flashed to a Presa TC 100 W?

Thank you and keep up the nice work!

Cheers,

Hans

ClockSelect, 2016-10-04:
Hello :) Thanks for your support.

In its current version, the firmware cannot be uploaded to the TC100W; and I don't know if it would work either. I must investigate. Since the battery management seems to be different, it may require some work.

ClockSelect, 2016-10-04:
Had a look to Wismec's website, the firmware for the TC100W is not online yet, so no investigation possible yet.

dabugen, 2016-10-04:
Man, you are fast with replying as well. Yesterday I had installed one version for my Presa TC 75 in the afternoon and there was already a new one in the evening, LOL. I love that:)

How about this, I will buy a Presa TC 100 W and capture the firmware for you and upload it?

dabugen, 2016-10-04:
Hmm, I just notice, there seems to be no tool to read the firmware, not even from the TC 75, right? Or do you possibly know one?

ClockSelect, 2016-10-04:
I have a way, but it's dangerous. It can definitively brick a box if something goes wrong.
Better wait for the firmware beeing online (can be long, unfortunately), or consider another 25mm box for which we have it - an eVic VTwo, for example.

dabugen, 2016-10-04:
OK, well, I get them for 24$ wholesale, worth a try for the features I could have on it:) Also, I read everywhere that even on that box, they say they recommend to use a max of 75 W. To me this all sounds like this is just the standard electronic of the Presa 75, with just the max wattage upped to 100. What tool would I need to use to read out the firmware? The other way around, I could also try to just flash the My EVIC firmware to it and see what happens. Which way is "less" dangerous you think?

ClockSelect, 2016-10-04:
When you have it, use the NFirmwareEditor (or whatever) to extract the dataflash parameters, and we'll look at it.

ClockSelect, 2016-10-04:
(BTW, by looking at Wismec's site, I saw they had a new RX75 I missed. I added compatibility for it in the meantime).

dabugen, 2016-10-04:
OK great, I´ll let you know. But wondering how it is for the VTwo, is it the exact same firmware as on the Evic VTC Mini, just with the upper wattage of 80 instead of 75?

ClockSelect, 2016-10-04:
Yes; wattage is upped to 80W just because of the bigger battery.
But firmwares are not *exactly* the same; there are subtle details, such as shunt resistance value, that vary from box to box and hardware versions. Also, some boxes do not read the battery voltage in the same place, etc. Little subtle differences.

dabugen, 2016-10-04:
Interesting, OK, well, I will get back once I got the TC 100. Thanks for the great work!

dabugen, 2016-10-11:
It still didn´t arrive yet, but is on it´s way says "China" :)

ClockSelect, 2016-10-12:
Great! Just a few more weeks^^

dabugen, 2016-10-12:
More like days now, shipping never takes longer than ~10 days from China -> Europe from the vendor I always use.

ClockSelect, 2016-10-13:
Oh, to answer one of your questions I missed: don not try to force flashing of the myevic firmware on the box. It may work, but it'll work badly, if ever.

dabugen, 2016-10-13:
Hehe, OK, I will just let you know once I have it here so that we can walk through it. You said there is a way to possibly read the firmware and then you could possibly adapt My Evic to it?

ClockSelect, 2016-10-17:
Yes, there is a way.

ClockSelect, 2016-10-26:
Wismec released there firmware, today^^
So me too.

@dabugen : Did you receive your box?

dabugen, 2016-10-26:
Wow, great stuff to hear that! No, it did not arrive yet, stuck at custom duty this time, but should be only a few days from now. I will report back. Great work!!!

ClockSelect, 2016-10-26:
It's been tested by a French forumer; seems to work fine.

dabugen, 2016-10-26:
Very good!! I will reply as soon as I´ve got mine, eager to test it. Thank you!

ClockSelect, 2016-10-29:
So final answer is: Yes, it will.

closed 2016-10-29 by ClockSelect

dabugen, 2016-11-11:
Just wanted to update that it has been working flawlessly for over a week now! Thank you for the good work.