ClockSelect/myevic #61: Shunt resistance?

dabugen, 2016-10-04:
Hi :)

I was wondering if someone can help to define what the shunt resistance actually is. Is it the resistance of the actual hardware circuit that is always there and hence all measured resistances are shifted by that value, like a permanent bias?

Thank you:)

ClockSelect, 2016-10-05:
Hello
I'll try to explain.
Map is:

bat.gnd- -> ato -> shunt -> bbc -> another bbc -> bat+

In fact, the firmware can only measure voltages; it needs some resistance to do that. So any measure done by the firmware is done through a resistance, somewhere.

If the FW wants to measure current or power on the atomizer side, it has to do so through a resistance called the "shunt". It's a useless resistance, whom purpose is to print a voltage given some current.
In all boxes, a (not-so)very small resistance, around 100 mOhms, is plugged in serial with the atomizer for the FW to read something.
It's purpose is to not mess with usual usage, while providing the information. And it works, more or less.

ClockSelect, 2016-10-05:
Maybe you wonder if editing your shunt resistance will improve your vaping experience; answer is: NO.
If you build your coils, and see some systematic discrepancy between your ohmmeter and the box, then maybe, your shunt may be in fault; but it's an unusual case. Maybe your ohmmeter is in fault.
Anyway, if you ask yourself if: NO.

dabugen, 2016-10-05:
Thanks for the informations, so pretty much what I thought, it´s a kind of bias for the resistance measurement. No, no need to edit it, my offset is fine and measurement spot on. Just was curious in general how exactly it works.

Thanks a lot:)

ClockSelect, 2016-10-05:
Please :) This question will help others, I think.

eugenesan, 2016-10-05:
That Q/A reminded of a thought I had in the past...

I've noticed that resistance measuring on my eVic is not consistent.
For example if I remove the atomizer at evening and then reinstall it at the morning, the resistance is almost always different than before it was removed.
I assumed that the reason is thermal/mechanic changes in the atomizer but I wasn't entirely sure since at same time other atomizers would exhibit similar resistance differences.

I wonder, what if resistance inconsistency is caused by thermal changes to the shunt?
Maybe we could try to "relate" shunt constant to the board temperature?

ClockSelect, 2016-10-06:
The shunt is not a thermistor nor the board temperature varies widely - does not exceed 40° in normal use. It shouldn't show discrepancies above the second decimal place. Also, I find board temp measurement rather inaccurate. I think it'll cause more troubles than it solves issues.

eugenesan, 2016-10-06:
I see.
Thanks for the explanation :+1:

closed 2016-10-14 by ClockSelect