ClockSelect/myevic #62: ml/Day display

dabugen, 2016-10-06:
Hi:)

I think the ml/Day feature is a dope idea:) However, just letting you know that I´ve vaped 5ml today already and it shows average of 0.40ml/day right now, which is way to far off. I know it´s experimental, but if you need any kind of input from users, let me know, I am ready to help:)

Cheers,

Hans

ClockSelect, 2016-10-06:
Hello

I known it doesn't work very well yet. And I don't know if it can reliably work at all, that's why it's experimental.

What is your setup? Coil, vaping mode, etc.

Also, if you changed your battery during the day or reset your box by any mean, you'd have restart from zero. Value is not saved yet.

ClockSelect, 2016-10-06:
See issue #51 for discussions on the subject; I'll close this issue as duplicate.

closed 2016-10-06 by ClockSelect