ClockSelect/myevic #64: lock VTC Mini eVic after 10-20 puffs ? possible?

bombek5, 2016-10-06:
whether it is possible to do in the firmware, the lock mode after a certain of number of puffs (or alternatively after a certain count of time floating, 10 minutes for example soar - 30 min lock)
(Number of puffs or steaming time as the period of the lock mode is set in the menu) and on the main menu, while the lock by pressing the Fire button will lock the countdown time)
that often do not smoke

falconettt, 2016-10-06:
What is your goal? 
You might want to obtain variety this way? Yes, you'll get it this way.

If you want to control vaping - NO.  I already went this route.  Lock mode after a certain of number of anything will not replace will power. You'll end up with longer inhale time and higher wattage.
IMO the only way to control is to align "how much liquid you vape a day" with the desired.

ClockSelect, 2016-10-06:
Agree.
I'm OK to display information; after that, user is responsible for using it as [s]he wishes.
After all, if you set up your box to enforce a limit, you're always free to remove the enforcement. So what's the point, really?
A good option would be a beep warning, but there's no beep on this box.

dabugen, 2016-10-06:
Makes no sense to me as well, the ml/Day is a good ballpark figure though to control yourself, if it can be made somewhat more accurate. But apart from this, you have to have some will to control your vaping, just like with anything in life.

MicroSur, 2016-10-08:
I patch official firmware to turn mod off after a few puffs (and after amount of puff time), very useful.

bombek5, 2016-10-09:
how to do it? can your patch?

MicroSur, 2016-10-09:
All my work about custom FW is here: https://www.dropbox.com/s/83zd19gu05pl3r6/EvicVTCFont.rar?dl=1
This patch is for VTC mini 3.02 (old) and Vtwo mini 4.02

ClockSelect, 2016-10-19:
Wich patch is it? And how do I read paches? I mean: read, not apply.

MicroSur, 2016-10-22:
I am sure you can implement this feature without realizing how I do it with patch )
I just said that this is a useful feature and I use it constantly (puffs_off_mod.patch for 3.02 vtc mini).

And please change the link from FWUpdater to VTCFont in readme, updater is a part of my software. No more direct link to FWUpdater.
https://www.dropbox.com/s/83zd19gu05pl3r6/EvicVTCFont.rar?dl=1

Villista, 2016-11-03:
Good time to all! Really a very important addition to this project! Many people try to quit smoking using e-cigarettes. However, the electronic cigarette is not a factor of smoking one cigarette end! And vapor continues nipples mode! Reminder or lock would be a very great help for these people. And binding to the volume of the vapored liquid (ml) would be more correct!

closed 2016-11-16 by ClockSelect