ClockSelect/myevic #65: Unnecessary "%"

eVic75, 2016-10-06:
Is this "%" necessary?
[image]

Best regards

dabugen, 2016-10-06:
It means "battery percentage" and hence Batt% is absolutely right.

eVic75, 2016-10-06:
I know what it means :) It WAS absolutely right when the options were:
**Batt% Of
Batt% Off**
Now, when the options are:
**Batt: %
Batt: V
Batt: Off**
"Batt%" looks weird...

ClockSelect, 2016-10-06:
Seriously... You opened an issue for this^^
I'll remove the %.

eVic75, 2016-10-06:
Sorry, I didn't know how to let you know about it.

ClockSelect, 2016-10-06:
Don't close the case before it's done or I may forget^^

eVic75, 2016-10-06:
Sorry again. It isn't my good day :)

ClockSelect, 2016-10-06:
Don't mind. :)

dabugen, 2016-10-07:
It´s a SERIOUS problem, we should open 2 threads about it actually. My Evic almost BURNED because of this bug!!!!!!

ClockSelect, 2016-10-08:
Finally, I didn't remove the %.

closed 2016-10-08 by ClockSelect