ClockSelect/myevic #66: Wrong voltage/percentage battery

GittyHub22, 2016-10-07:
Hello, 

Just charged my battery external, it shows 4.06v and on my charger it shows 4.23v
Also my battery is a 30Q but if i use that setting in the expert menu the percentage is off. 
And if I turn my device off (Presa 75w) and connect a charge cable it shows a different percentage than when the device is turned on, no matter what setting.

And these ones are extremely minor but the cube screensaver does not work anymore since yesterday's updates I believe. I know the ml vape is experimental but that does not seem to do anything.

I hope you can do something with my findings, but you have already done a great job.

Regards,

NikitaAtamanchuk, 2016-10-07:
Hello! I also have a problem. My HE4 in a build of 161006 is charged to 4.24v or higher. I unplug the charging, because I'm afraid to break the battery. Before installing this build, battery exactly charging to 4.2v

ClockSelect, 2016-10-08:
@NikitaAtamanchuk: what's your box?

@GittyHub22: Id did not change anything in battery measurement from the genuine Presa75 firmware. Unfortunately, I do not have a Presa by hand to test; what I'd like is you to reflash the original firmware and do the very same tests about volyage, and tell me on each case how I'm off.

dabugen, 2016-10-08:
All fine with my Samsung 30Q´s and my Presa 75 TC here with the voltages.

ClockSelect, 2016-10-08:
In theory, if the voltage measurementis ok, the % should be too - Percentage curves are rather fine.
But different boxes may probe voltages differently; for example, the VTC-Mini measurement is systematically off by -0.05V. So I apply a correction to it, but only on this box, not on the Presa. I'd like to know how different boxes are off on this measure, to have to correct offset for each box.

dabugen, 2016-10-08:
Could I help with that with my Presa 75 TC somehow?

NikitaAtamanchuk, 2016-10-08:
@ClockSelect eVic VTC Mini

GittyHub22, 2016-10-08:
The orig firmware has no percentage or voltage, only patches with mfirmwareeditor and i noticed something in nfirmwareeditor device configuration: the battery is set as HE2 (85%) even though the device shows GEN in the expert settings and even if you change it in device configuration, the next time it will still show HE2 (or GEN is the expert settings)

ClockSelect, 2016-10-08:
@dabugen: It would be great if you could compare the battery voltage shown by the box with the real voltage given by an external voltmeter, if you have one.

@NikitaAtamanchuk: voltage rises when charging, and immediately drops when you stop charging, that's perfectly normal. 4.22v is a very high voltage; not too much to be a concern, but a charger should not go so high.

NikitaAtamanchuk, 2016-10-08:
My evic always charge up to 4.2v.
As far as I know, the li-ion battery needs to be charged to the level of 4.2v.

ClockSelect, 2016-10-08:
I see. On my VTC mini, the voltage reading of the battery is systematically off by -0.05v; so I added a corrective offset of +0.05v so my box now reads the correct voltage (checked with an external voltmeter).
But it seems that not all boxes are made equal on this regard.
Maybe I should make this configurable.
Anyway, don't be afraid for your battery; it's just the way the number is displayed. The actual voltage sent to the battery does not change from before.

NikitaAtamanchuk, 2016-10-08:
I'm understood, thank you.
By the way, is it possible to change the charging current? At a current of 500 mA charging is very long. Li-ion batteries can be charged at currents up to 0.5C (for HE4 is 1250 mA)

ClockSelect, 2016-10-08:
Nope; 500mA is the limit enforced by the USB protocol.

NikitaAtamanchuk, 2016-10-08:
wiki: USB 3.0 increases the current limit from the levels defined by the USB 2.0 specification. Current draw for low-power SuperSpeed devices (ones that are limited to 1 unit load of power) is now 150mA, an increase from 100mA defined in USB 2.0. For high-power SuperSpeed devices, 6 unit loads may be consumed, for a limit of 900mA, 4.5W (4.5 watts), an increase from the limit of 500mA defined in USB 2.0.

ClockSelect, 2016-10-09:
Agree, but the USB circuitry of the MCU is a USB 1.1.

ClockSelect, 2016-10-09:
I reset the battery voltage offset correction to zero, and added an option to fix it by hand.
(BVO in the Expert menu).
It should fit everyone :)

NikitaAtamanchuk, 2016-10-09:
Great, thanks!

GittyHub22, 2016-10-09:
Still the same issue, but now I know that the voltage showing when off is correct (I measured the battery), The settings in the expert menu (GEN or 30Q) only change the percentage for some reason. And if I change BVO the voltage showing when off and charging is wrong. The difference between off/on is 0.07v (BVO will only let you change max 0.05v but again: the wrong voltage is still showing when off).

ClockSelect, 2016-10-09:
What do you mean by "when off"?
Yes, the expert menu setting only changes % calculation.
When you're charging, you're supposed to have a voltage rise by about +0.05, it's perfectly normal and due to the charging process.

GittyHub22, 2016-10-09:
When the device is turned off.  Not an expert in batteries, but how come dabugen does not have this issue then?

closed 2016-10-09 by GittyHub22