ClockSelect/myevic #67: VTC Mini - CLK Speed?

jaaxx, 2016-10-08:
The instructions say to use + & - to adjust the time at the top of the screen, but on my VTC mini it only adjusts the ratio on the center of the screen.  User error on my part?

eVic75, 2016-10-08:
Do you mean: Menus > Clock > Clock Speed? I think you have old version of software.  Install newer/newest.

[image]

ClockSelect, 2016-10-08:
If you don't see the time changing on top by adjusting the ratio, it means your box did not sleep long enough the last time it slept. It may be due to the fact it slept a few seconds just before you entered the menu. When you wake up your box after some time and wish to adjust the ratio, it's important not to let it enter sleep mode again in-between.

closed 2016-10-14 by ClockSelect