ClockSelect/myevic #71: Cloud Volume mode - Energy based vaping

Rudgas, 2016-10-10:
I think it would be interesting to add a mode where you can input the max. temperature/wattage and select a certain amount of energy in Joules which may be expended per draw. So wheather you want a slow draw or fast draw the amount of vapor produced would stay the same.
I'm thinking of this in the context of either juice-flow control or as my interest is in flavour control. Some liquids are just too intensive for a full lung draw. Also sometimes I think in VW mode flovour suffers from over-heating and TC mode doesn't seem that accurate.

ClockSelect, 2016-10-10:
There's a way you can do that right now: configure the puff duration limit (Vaping>Protec) according to your power.
Watts*Protec = Max Joules.
I know it's not ideal, but at least you can test if it's really what you want.

dabugen, 2016-10-11:
TC is very accurate with titan for me and gives the best flavor if you configure it right.

Rudgas, 2016-10-12:
Thanks for the quick reply!

closed 2016-10-22 by ClockSelect