ClockSelect/myevic #72: ml/Day not working anymore?

dabugen, 2016-10-11:
I am wondering that with the updates of the last days (also the current one which I´ve just downloaded again), the ml/Day always stays at 0. Is that normal because the feature is being reworked or something? Thank you.

ClockSelect, 2016-10-11:
Strange; it still works on my side.
Does the simple display of quantity in mL works?

dabugen, 2016-10-11:
Hi:)

I have ml/d as the third line (Pwr, Coil, ml/d) and it always shows:

0.00 ml/d, regardless how much I vape.

I am in TC M1 mode, does that make a difference?

Thanks:)

ClockSelect, 2016-10-11:
No, Vaping mode is irrelevant.
What does it shows if you set the display to ml instead of ml/day? Still zero?
Maybe you did not reset value for too long; ATM, you're supposed to reset when you wake up.

Daveee10, 2016-10-11:
I can say what ClockSelect sad... It works correctly...

dabugen, 2016-10-11:
If I change it from ml/Day to just ml, right now it says 2.66ml and increases while I vape on it. When switching back to ml/Day, it again shows 0.

Daveee10, 2016-10-11:
I think you need to reset it. When the ml/day selected and you changing the third line info, when blinking the roy press and hold fire for couple of sec. That's all..

dabugen, 2016-10-11:
No luck. It resets to 0 indeed (ml mode) and if I start to vape the ml increase, but ml/D mode still shows 0 and won´t count up as soon as I start  to vape (but the ml mode does!).

ClockSelect, 2016-10-11:
Indeed, reset rests. It's its purpose. Vape and see.

dabugen, 2016-10-11:
I am sorry to say, but while ml counts up (now 1.00 ml already), ml/D stays straight at zero still.

falconettt, 2016-10-11:
May be good idea to upload original firmware. 
And then upload myevic firmware again.
2 min work, yes?

dabugen, 2016-10-11:
Did already, no change.

falconettt, 2016-10-11:
I can say what ClockSelect and Daveee10 sad... It works correctly...

dabugen, 2016-10-11:
That´s great for you guys, but doesn´t fix it on my Presa 75 TC either. No worries, I will just not be using the feature, it´s not that important at all...

dabugen, 2016-10-11:
I just did a complete reset of the dataflash now via NFirmwareEditor, all settings did reset, but ml/D still does not count anything, only ml does. (Wismec Presa TC 75 in TCR M1 mode)

t2dYuri, 2016-10-12:
When i installed the latest firmware version, "ml/kJ" value in menu was "0", so there was always zeroes in third line (ml & ml/d) counters too...
After I changed the value to non-zero, counters started to work.
Maybe it was better to set any non-zero default value in "ml/kJ"?
// _eVic VTwo Mini_ //
Thank you. Great job

ClockSelect, 2016-10-12:
It's may be due to some data I had to move around in the dataflash for better OFW compatibility, so it may have messed up some parameters.
Anyway it shouldn't have been zero since it's checked for reasonable range at startup.
Anyway, you've fixed it^^
Default value is 360.

ClockSelect, 2016-10-12:
Found the bug, value was set to zero on a dataflash reset.

dabugen, 2016-10-12:
My value was at 360 even after the data flash reset, but it´s still showing 0 ml/Day for me right now.

dabugen, 2016-10-12:
I just flashed your firmware from 12 minutes ago, still no change, "ml/D" stays at 0 all the time, while "ml" counts up fine.

Apart from this, when I flashed this firmware, the box first went black and won´t turn on. Luckily NFirmware Editor still connected to it, but showed software version as 0.00 and "LDRom" as boot option. Luckily it flashed the firmware again and now it´s working again. What the heck was that?

ClockSelect, 2016-10-13:
I happens. Firmware flashing is not 100% reliable. But you reflash and it comes back.

dabugen, 2016-10-13:
OK:) So what´s up with that ml/D display?

dabugen, 2016-10-13:
I just wanted to add again: on previous MyEvic versions (~6 days ago) this IS working fine with the ml/D.

dabugen, 2016-10-13:
Btw, I´ve just flashed your latest version from ~15 hours ago and had to flash 4 times now before the Presa 75 TC finally came back online. So the flashing failed ~4 times in a row now (removing battery / USB inbetween several times), always said "Boot: LDROM". On the 5th flash it then came back finally. What´s going on? Is my Presa "at the end"?

dabugen, 2016-10-13:
Should I stop asking about this problem and just not use it? Just asking....

dabugen, 2016-10-14:
LOL, are you ignoring me? I see that you post everywhere, except here.

dabugen, 2016-10-14:
By the way, I´ve just flashed MyEvic (the version from 2 hours ago) to a brand new Evic VTC Mini that has never been used before (just arrived today), yet the "ml/D" does NOT display anything at all, like on my Presa 75 TC. Only "ml" counts up fine, also like on my Presa 75 TC. So this must clearly be a bug.

Daveee10, 2016-10-14:
Set the time. Maybe...

dabugen, 2016-10-14:
Wow, a reply!

dabugen, 2016-10-14:
By the way, I found the fix for the steadily failing flash myself now, just turning off "X32" in the Expert settings and every firmware update goes fine, no more fails!

mirekwrozek, 2016-10-14:
I have similar problem, when I updated a firmware my VTwo mini is frozen. I must to flashed original firmware a few times(with battery and without battery) and flashed the newest firmware. I noticed that after flashed firmware and reset dataflash isn't cleaned liquid count.
sorry my english

dabugen, 2016-10-14:
Maybe you should also try to turn off "X32" in "Menu" -> "Expert"

ClockSelect, 2016-10-14:
Liquid count isn't stored in dataflash, but in the RTC registers, so it's nor reset with the dataflash.
I can't really see why X32 would cause issues with firmware flashing. Anyway, it's always better to turn X32 off if your box is known not to have built-in RTC.

dabugen, 2016-10-14:
No idea, but there was another thread here that mentioned this issue as well and turning off X32 helped there as well. Let´s see if I can find the link.

dabugen, 2016-10-14:
Found it: https://github.com/ClockSelect/myevic/issues/16

ClockSelect, 2016-10-14:
Yes, I remember that. But the X32 should normaly not prevent the box from working any more. I test it at startup and disable it if needed. Strange. Maybe some boxes without X32 manage to pass the test.

dabugen, 2016-10-14:
Yes, what I notice, that with X32 = on, the flash fails each second time as the box won´t turn back on and shows "LDROM" along with "Software version: 0.00" (tried 20 times in a row just to make sure, 10 times it failed). With X32 = off, each single flash of the firmware goes fine (also 20 tries), there is no fail ever.

dabugen, 2016-10-14:
What the heck is X32 anyway? :)

ClockSelect, 2016-10-15:
It's the 32.768kHz crystal used to drive the Real-Time Clock.
On box that do not have natively a RTC, the X32 is not even soldered on the PCB; but leakage currents may make the box believe there is one.
Also, the RTC feature may malfunction on some RTC-enabled boxes, so there is a need for this option to not use the X32.

ClockSelect, 2016-10-15:
Should prevent issues.

ClockSelect, 2016-10-15:
So I think we're done with that ml/d issue.

closed 2016-10-15 by ClockSelect

Justin-L-Franks, 2016-11-22:
On my Cuboid v1.01, the mL/d option does not work; it remains at 0.00 mL/d even after setting the clock, then resetting the counter as suggested in this thread. Switching to mL will work correctly (if I set it to mL/d, reset the counter, vape, then switch to mL, the value is > 0.00).