ClockSelect/myevic #73: Changing coil resistance won´t reflect instantly

dabugen, 2016-10-11:
Hi,

if changing the coil resistance via the menu (Fire + Plus) -> Coils -> Manage -> TCR (since I use that mode), it is not reflected in the main-screen right away, only after ~10 minutes or ~20 puffs it then changes to the new value I´ve set in the menu. It´s also easy to feel that it wasn´t changed, so it is not just the displaying.

Thank you:)

ClockSelect, 2016-10-11:
Yes, the menu changes the values in the dataflash but not the values bufferized by the firmware.
I'll have a look.

dabugen, 2016-10-11:
Ah OK, so it´s a bug?

Daveee10, 2016-10-11:
Yes. But you can't wait X minutes or X puffs... Take off the atomizer then take it on again..

dabugen, 2016-10-11:
Hmm, but can´t this be done without having to remove it? Would be much cooler :)

ClockSelect, 2016-10-11:
I said I'll have a look.

dabugen, 2016-10-11:
Sorry.. Ok, thanks.

ClockSelect, 2016-10-12:
Give it a try!

eVic75, 2016-10-12:
Fixed :) Works properly.

dabugen, 2016-10-12:
Great, it´s working fine now and changes instantly! Thanks!

ClockSelect, 2016-10-12:
Great :)
Closing.

closed 2016-10-12 by ClockSelect