ClockSelect/myevic #74: RTC Speed

Daveee10, 2016-10-12:
Hy. I tested your new RTC modification...
From yesterday night (22:00) to today morning (5:00) It works very well, only 1-2 sec of difference (Little faster than the clock on my laptop. BUT. Today the clocks shows 19:00 when the real clock is 18:47... I don't know how why. Maybe because today I'm using the clock on mainview or because when firing the oscillator not give constant value or because the clock correction value maybe not useless...
Have you got any idea?

ClockSelect, 2016-10-12:
No; I saw this when waking up on my VTwo Mini (I have a VTwo without functioning X32, so I set X32 OFF and the clock is emulated) but not on my Cuboid (also without X32), so it may depend on the hardware. And I wasn't using the boxes since I was sleeping. It has to be investigated. What surprises me is that it *advances*.
Clock speed ratio is not used at all ATM when Light Sleep mode is on.

Daveee10, 2016-10-12:
VTC Mini:
HW version: 1.0.0

Wismec Presa 75W

I'm using this two mod whole day and the problem is same..
I think the firing cause the problem.. Maybe we need compare two timer when the mod not sleeping and firing... IMO In the "sleep" mode the oscillator works well now.

ClockSelect, 2016-10-13:
Well, I found out something: had my two boxes perfectly synched and on time at wake-up today, and my VTC jumped instantly 34 minutes forward at the first puff. Cuboid did not. Then both boxes remained stable.

So the jump into the future seems to happen at the first puff after wake-up. It's a hint.

PaoloW8, 2016-10-14:
VTC Mini
HW Version: 1.00

clock works fine, after 2 or 3 days still correct (with LSL on) but every time I flash a new firmware version the clock goes forward about 30 min

ClockSelect, 2016-10-14:
@PaoloW8 : That's interesting. A reproducible thing.

PaoloW8, 2016-10-14:
I just retried now (with the same firmware file I used this morning) and the clock went ahead by only three minutes. after that I flashed again and the clock is not changed.
I never saw change the date and never saw the clock go backwards, only forward

ClockSelect, 2016-10-14:
When uploading firmware, clock jump is due to the reset command.
This *should* be fixed.
But I'm not sure it'll fix every possible clocks jumps.

Daveee10, 2016-10-14:
Since yesterday the clock works very well. I can't reproduce the clock jumps issue after night. Today morning it's good... The whole day works good too... After firmware uploading the clock jumps only 8 sec (forward)... IMHO this is a good result on a box without X32 crystal...

P.S.: I forgot to say the clock only jumps to forward but never backward...

ClockSelect, 2016-10-14:
Good news. Let's wait a day or two.

Daveee10, 2016-10-15:
If I change a battery the clock jumps forward 30 mins

bombek5, 2016-10-15:
I have the same , jumps forward 30 mins

ClockSelect, 2016-10-16:
Should no more.

PaoloW8, 2016-10-16:
Now it seems ok :)
no clock jump on flash firmware
no clock jump on battery swap

thanks @ClockSelect

ClockSelect, 2016-10-19:
Hope we're done with that.

closed 2016-10-19 by ClockSelect