ClockSelect/myevic #75: Awakening delay after sleeping mode

t2dYuri, 2016-10-12:
Hi again. I noticed a strange awakening-delay with the latest firmware after the device has been inactive for some time (about 3 minutes & more). When i press any button - there is a delay of 1-2 seconds before the display lights up. It's not a problem, but there is the same delay before firing, so i should press the button and wait 1-2 sec. before the first puff...
There was no such problem in my previous installed myEvic version (from Oct 1, 2016).
// _eVic VTwo Mini_ //
Thank You.

ClockSelect, 2016-10-12:
Strange; I do not experiment this. My delays goes from instant to barely half a second. I'd need more informations to investigate.

Daveee10, 2016-10-13:
This delay since merge the fasradc branch

ClockSelect, 2016-10-13:
And even since? I reverted almost everything concerning wake-up moment from this branch. Only remaining difference is an EADC_Open(). I removed it for test purpose (it's useless anyway) and see no difference.
My VTC-Mini wakes up instantly, we VTwo I used the last few days waked up instantly too, the Cuboid I'm currently using wakes up in fraction of a second...
Can't see, really.

dabugen, 2016-10-13:
I don´t dare to say, but I have the same issue. Sometimes, after being in sleep mode, my Presa 75 TC won´t even turn EVER again until I remove the battery for ~5 seconds, then I can power it back on with pressing fire 5 times as usually. I also didn´t have this with older MyEvic firmwares.

acevalsl, 2016-10-13:
hi, i have same problem with vtwo mini.

ClockSelect, 2016-10-13:
Do you have the same issues with the LSL option OFF in the Expert menu?

acevalsl, 2016-10-13:
Yes

t2dYuri, 2016-10-13:
In VTwo Mini LSL was OFF by default

acevalsl, 2016-10-13:
I have it off

ClockSelect, 2016-10-14:
It's gonna be hard since I can't reproduce.
@Daveee10: Are you sure it's since this commit? I can't see in the diff why it would cause this issue.

acevalsl, 2016-10-14:
i am not secure...... i think since no the last, one more..
in this moment i have old version and work fine, but if i use one more new like 161006 or this not work
sorry my english

ClockSelect, 2016-10-14:
@acevalsl : and what version are you currently using? so we have the lower bound to search.

acevalsl, 2016-10-14:
Now I can not see version number, but is one before 161006

ClockSelect, 2016-10-14:
When you set the USB option in the Expert menu to "DSK" to use the box as an USB disk, what is the date/time of the FIRMWARE.BIN file in the drive?

acevalsl, 2016-10-14:
Excuseme, but I am in hollidays and I don't have access to PC. Sunday return.
Sorry

acevalsl, 2016-10-14:
In my PC, bin file say 26 sept.  I see with teamviewer. I used this file for vtwo and this work
[image]

Daveee10, 2016-10-14:
@ClockSelect: The issue come with fastadc but I saw how you removed everything.

ClockSelect, 2016-10-15:
If you revert to the manufacturer's firmware, does the problem disappear? (ovbiously, it should)

acevalsl, 2016-10-15:
Which is your hardware version?

t2dYuri, 2016-10-15:
Rolled into the original fw. v4.04 - instant response from any box state.
[image]
After that I tried to install the latest (today's) myEvic -- got "Update completed" and dead box (no reaction to any actions)
[image]
Pulled battery off and tried to install myEvic again - got the same result... so I have to work with native firmware for now, but I loved the "ml" counter which allowed me to use non-transparent tanks :(

ClockSelect, 2016-10-15:
Yes we had some firmware update issues this week-end...
But a dead box is strange. Shouldn't occur.

falconettt, 2016-10-18:
Sorry to say,
Just uploaded new firmware, and have awakening-delay after the device has been inactive for some time.
It's not always but sometimes. 3-5 fire clicks make box alive again. LSL change doesn't help.

acevalsl, 2016-10-23:
Hi, I'm testing last software and delay is missing.
Please test it for answer..........

t2dYuri, 2016-10-23:
Really. With the latest  myEvic updates I have an instant response too.
Thank You.
// _eVic VTwo Mini, HW 1.00, TC Ti_ //

ClockSelect, 2016-10-23:
That's great, but... *Why* ???
I'd really like to figure out the point.

acevalsl, 2016-10-24:
resistance measure??

falconettt, 2016-10-24:
I did some research and found the moment of occurrence. 
By the way, it is not a delay, box just shuts down under the following circumstances:

"Long fire protection" - LFP.

So you inhale till LFP triggered, see this annoying "LFP screen" but do not touch anything just put your box aside. Screensaver works and mod goes to sleep.  At this moment you have mod switched off (why???).
You can power it back on only with pressing fire 5 times. 

If LFP is not triggered you will not get this "delay".

ClockSelect, 2016-10-24:
Yes, box switches off 10 seconds after after a long fire without response - it's on purpose, it considers it's in your pocket.

But switching off, requesting five clicks to wake up, and having a wake-up delay are two different things.

acevalsl, 2016-10-25:
I never need press 5 times for wake up

ClockSelect, 2016-10-25:
You don't have a Presa ^^

ClockSelect, 2016-10-26:
I think I found the *why*.

acevalsl, 2016-10-26:
??????

ClockSelect, 2016-10-26:
Seems closable.

closed 2016-10-26 by ClockSelect