ClockSelect/myevic #76: Cuboid real time resistance and voltage of both batteries in third line

rocknrollman, 2016-10-13:
Hey) Delivered your firmware on a cuboid. It very cool. Everything what i dream that: display of resistance in real time in thirds line, like in evic and that in it the voltage of both batteries was displayed. It is possible to make it? I will be very grateful)

ClockSelect, 2016-10-13:
The resistance in real-time with a milliohm accuracy on third line is already available.
And if you look closely, you'll see that the coil on second line is *also* in real-time.

For the both batteries: it's in the TODO list, but I don't really consider it as urgent - What you see is the voltage and level of the *lowest* of the two batteries. As long as you do not have an imbalanced warning, you know that the second battery is at most at +0.02V from the first. Is it really worth displaying two identical numbers?

rocknrollman, 2016-10-13:
In an eviс just resistance is updated with a bigger frequency than in a cuboid, here I also want effect as in an eviс. And concerning batteries, here already on your discretion, for me it is just a habit :)

ClockSelect, 2016-10-13:
They're updated every tenth of a second when firing, and every few seconds when idle, on both boxes.

ClockSelect, 2016-10-13:
I improved the refresh rate of the real-time rez, maybe it will help^^
I agree it's better.

ClockSelect, 2016-10-14:
I'll keep this issue open since I think many others will ask for dual-battery display.

rocknrollman, 2016-10-14:
Thank you, resistance now is perfect!

bluenazgul, 2016-10-20:
another feature for 2 Battery Mods would be to Diplay the two battery at the bottom line like in original Fimware

But for the Theme Creators it would be easier to keep the single icon

closed 2017-01-21 by ClockSelect

nbraunstein, 2017-01-24:
Is two battery display possible on the rx 2/3?

ClockSelect, 2017-01-24:
No, battery display is unified.
All batteries are supposed to be between 0.3V between the lowest and max voltage; if not, a visible balance warning is issued.
You can always consult comprehensive battery voltages in the Expert>BVO menu, or by switching the box off and holding Fire-<->.