ClockSelect/myevic #80: Please confirm the Hardware version

batsteek, 2016-10-15:
No issues at all since the very beginning, and for the very last commit (f92ddb0d81fd60eb7c4f6ddaa5abfceb72054ce3) I get the following message when I try to update the box: Please confirm the Hardware version.

Any idea?

https://www.dropbox.com/s/gv0oqjyudiy2am3/Capture%20d%27%C3%A9cran%202016-10-15%2009.57.54.jpg?dl=0

ClockSelect, 2016-10-15:
Yes; you lost the hardware version.
I happens when the updater fails to correctly rewrite the dataflash to reset the box during the update process. It's not related to the firmware version.

To recover it, you have to remember what was your hardware version; if not, you'll have to try starting with 1.00.

Follow the following guideline:
* With an utlity such as [NFirmwareEditor](https://github.com/TBXin/NFirmwareEditor/releases), download and save your dataflash to a file (if you're using NFE, it's in the Tools>Firmware Updater>Advanced Tab menu).
* With an HEX editor, remplace the first 8 bytes of the file by:  
```00 00 00 00 64 00 00 00```  
  The "64" should be your hardware version number in hexadecimal:  
  1.00 : 64  
  1.01 : 65  
  1.02 : 66  
  etc...  
* Rewrite the file to your box with the utility.

It should work.

If if works but you have a scrambled screen, it just means you have the wrong hardware version. Retry with 1.01 or 1.02.

Btw, what's your box model?

batsteek, 2016-10-15:
Thanks for the procedure. 
My box is an Evic VTC Mini, as for the hardware version, I don't remember so I'm going to try as per your explanations.

batsteek, 2016-10-15:
mmm ...
1- Regarding the Hardware version, if it's lost, I still see it from the Update utility & NFE and it says 1.00.  Still you think it's lost?

[image]

[image]

As for NFE 5.1, when I try to read/save the dataflash, I get the following message: "An exception occured during dataflash reading. Operation timed out".

[image]

Note: It's a Windows 10 on a Parallels Virtual Machine on my MAC. (and I never got any issues until today.)

batsteek, 2016-10-15:
Don't know why/how, but I finally managed to grab the dataflash and ... I see the hardware version (64) so ... I don't know what to do now ...

[image]

slavyk, 2016-10-15:
Maybe this help you [https://github.com/ClockSelect/myevic/issues/2#issuecomment-240336605](url)

slavyk, 2016-10-15:
maybe this comments help you [](https://github.com/ClockSelect/myevic/issues/2)

batsteek, 2016-10-15:
thanks @slavyk 

Okay okay!
Issue with my box is only with the last commit (f92ddb0d81fd60eb7c4f6ddaa5abfceb72054ce3)
I tried the previous one without any issue.

I hope it will help you :)

ClockSelect, 2016-10-15:
I see you have the NFE option ON. Set if to OFFto update firmwares.

batsteek, 2016-10-15:
Yep, this is what I do.
It was enabled only because I was dealing with NFE at the moment of the screenshot so (I guess) it's not the reason of my issue.

ClockSelect, 2016-10-15:
Found the issue.
Working on it.

ClockSelect, 2016-10-15:
gcc compiler "optimized" product IDs in the code by computing some from the others, so some boxes ID didn't appear any more in the binary file.

batsteek, 2016-10-15:
Still the same initial issue with the last commit ebac1b596ddc38f957d766d6878939a2a9eca667
-> Please confirm the Hardware version.

Hotix71, 2016-10-15:
ebac1b5 working for me with VTC, thanks!

Iam using NFE for the update....

ClockSelect, 2016-10-15:
Seems there's still an issue with official updater.
But NFE works.

ClockSelect, 2016-10-15:
Should work fine now.
Seems that some updaters want the product ID to be aligned on a 32-bits boundary.

batsteek, 2016-10-16:
And yes, it works now :)
Thanks for such a cool work!

closed 2016-10-16 by ClockSelect