ClockSelect/myevic #82: eVic VTwo Mini: Dead RTC or something else?

t2dYuri, 2016-10-16:
Hi. I didn't noticed this in-time and it does not apply to the ticket about the response delay, so here's my problem:
Yesterday I had trouble with one of the latest versions of myEvic - box transformed into brick during atomizer switch (just black screen and no reaction to any button clicks)...
So I connected it to PC without battery and flashed original 4.04 firmware. Everything became OK (as I thought). In few hours I tried to install the latest myEvic, got "Update successfull" and instantly bricked box again...
Rolled back into 4.04 as at first time... Today I saw a new myEvic version (with fixed fw. update) and tried to flash it -- got brick... also tried few older ones (from 09 oct, 01 oct, even one of september's commits) with the same result...
And after the last 4.04 flashing I noticed that my RTC showing wrong time & date (00:00 - 0000/00/00), tried to set it, but unsuccessfull - all the numbers remain zeroes :(
Downloaded NFE, backed up & reseted dataflash, tried to flash another version with this utility - no difference. Always see "Update completed" and brick on any of myEvic versions with empty fields in updater's window, or working original firmware, but with dead RTC...
Can I somehow still try to fix this problem?
Thank You.

mirekwrozek, 2016-10-16:
I have the same problem but my evic VTwo shows time. I can't change time value e.g. 1:40 when i clicked a few times - no result, still is 1:40 on all original firmware. Can you add firmware to reset box to deep default firmware(if will be bug everyone can recovery eVic to default firmware)?

ClockSelect, 2016-10-16:
There is no such thing as "deep default firmware" ^^
Once you've flashed the original firmware and reset the parameters, you're done.

Defective clocks are a known issue on VTwos. You're lucky enough not to be bricked with the original firmware.

I made a little debugging feature: try to reset your box (with latest commit) while holding the (-) (left) button; this will skip the RTC initialization part altogether and avoid using any of the RTC features.

If it works, you may try to change the X32 (OFF) and LSL (ON) settings in the Expert menu to see if it helps.

t2dYuri, 2016-10-16:
Hi. I didn't understand the firmware reset process exactly, so i just flashed the latest myEvic, box got bricked, i pulled-of battery, & connected USB while holding "-" btn... saw that screen lights up... switched-off USB, inserted battery and pushed "-" again for some time...
Anyway, myEvic is working again. Switched off X32 & LSL (i don't need RTC here), so i'm happy for now :)
Just want to ask a few questions:
- As i understood, native RTC is dead?
- How exactly i should update myEvic in the future? 
Thank You.

Solymo46, 2016-10-16:
Hi, i have the same probleme on Vtwo mini : when i try update for myevic.bin, brick. I go back to original 4.02 but now i can't flash myevic.bin and clock is on 0:00. 

I try your update but it don't work...

mirekwrozek, 2016-10-16:
I don't understand why last day everthing works good, but after update ([#74] Fix RTC jump at battery swap) my RTC is dead? Does the firmware damage some interior structure motherboard or emmc?

ClockSelect, 2016-10-16:
@t2dYuri : I can't really know if your RTC is dead, it would require more investigation. You will have no issue upgrading in the future as long as you keep the X32 OFF. Upgrade like you would normally. Just remember you have the Minus button option in case of problem.

@mirekwrozek : if something could have damage anything (and I can't see why, really) it would more likely have been before that fix than after. As I said before, I already saw this RTC issue arise with sole original firmware (see Joyetech's forum and other places), so I don't think I'm more "dangerous" than OFW. I use the hardware the same way.

@Solymo46 : remove USB cable, remove the battery, and hold the left button while plugging in USB or replacing the battery. When releasing the button, the box should come back to life.  After that, set the X32 option "OFF" in the Expert menu.

I'll try to get a box with this issue and I'll investigate deeper.

Solymo46, 2016-10-16:
I try with "-" but it don't work. So i use "+" for flash stock firmware and it work but not the clock. 

I try many time to flash your firmware, and now it's ok, time and all othres things. But now i can't revert to stock, it brick the box. 

Thank for your answer and sorry for my poor english...

mirekwrozek, 2016-10-17:
I noticed that when I get off battery(plugged only usb) on original firmware RTC works, but when I get in battery RTC is frozen.

ClockSelect, 2016-10-17:
RTC registers are bare user memories; they have no impact on the RTC working or not.

mirekwrozek, 2016-10-17:
So I don't know :/ Only all works fine when battery is out on official firmware(on different don't work with/without battery).

ClockSelect, 2016-10-17:
Boxes should not freeze anymore, whatever the RTC state (hopefully).

ClockSelect, 2016-10-17:
Also, BTW: I saw in the VTC-Dual code that Joyetech devs have written some freezing-prevention code for the RTC. They didn't do that in the original VTwo. They had the same issues, obviously.

Solymo46, 2016-10-19:
Hi, the last update work fine, can't revert to stock, but with your firmware, it work very good ! Thank !

closed 2016-10-22 by ClockSelect