ClockSelect/myevic #83: Editable TCR value for TC (NI / TI / SS) presets

t2dYuri, 2016-10-16:
Hi. I think it would be just great to give the ability to users to adjust TCR for each of default TC presets (with "reset-to-default" function).
Let's imagine a situation: I Have 2-3 atomizers, which runs at different coils: UD SS-316L (TCR ~ 120) few other SS-316L wires with TCR about 95-110, UD Titanium wire with huge TCR ~ 470... TC with UD coils needs Temp. above 300 deg, so no one of my wires don't match the default TC presets & I'm forced to use all my tanks with TCR M1-M3.
Thank You

Daveee10, 2016-10-16:
That's not a bad idea. BUT. The SS316 works with fixed TCR but the Ni and Ti wires works with another method in FW.

I was made a modification long time ago for 5 programabble TCR value this reason, and I want to make some other Temp mods for SS304, SS430, SS410, Dicodes Resistherm... Maybe @ClockSelect will make his magic with your idea 👍 
Another thing: Maybe the good ide how we can rename the TCR memory names...

ClockSelect, 2016-10-17:
I must have a closer look at how Ni and Ti TCR work cause I didn't do it yet.
I find that Ni and Ti presets work well with factory coils (they've been made to match, anyway).

That's a not-so-bad idea, it would close the SS TCR debate. I'll have to think how to do it. TCR Set menu will have to be rewritten.

Daveee10, 2016-10-17:
I will create some fixed temp modes and will make a pull request

ClockSelect, 2016-10-17:
What do you mean by "fixed temp modes"?

Daveee10, 2016-10-17:
Seems Temp Ni, Temp Ti and Temp SS316...
I will make SS304, SS410, SS430, Dicodes Resistherm

ClockSelect, 2016-10-17:
I'not really sure it's the way to go.
Or there is some revamp to do in the way modes are managed.
The first three modes should be collapsed in one with a submode, and we could act on the submode. ATM, Ni, Ti and SS are three distinct modes. Should not.
Then use the selectable modes flag in the DF to hide all additional submodes by default.
Some work.

ClockSelect, 2016-10-17:
Editable TCR presets seem simpler; who uses 4 different brand of SS anyway?

Daveee10, 2016-10-17:
I'm using different wires for my atomizers.

ClockSelect, 2016-10-21:
That's a first step.
Should be fine for 99% of users, imo.

closed 2016-10-31 by ClockSelect