ClockSelect/myevic #85: USB disconnect on sleep mode

fonix232, 2016-10-17:
The title makes it quite obvious. It is not an issue per say, but it would be nice if there was an active VCOM/HID connection, then the board would not go into deep sleep - keep the connection alive. In theory it shouldn't use much power, given how the device is already externally powered.

ClockSelect, 2016-10-18:
Why would you want that?

fonix232, 2016-10-18:
For tinkering, mainly, testing out stuff. I was trying to communicate with my Cuboid, and it kept disconnecting if I didn't set the sleep timeout higher or kept pressing the button, which was quite annoying.

ClockSelect, 2016-10-18:
Try to set Screen>Protection>Saver to "Off", and Stealth mode to Off also (Fire+Left).
The box should no longer enter deep sleep mode while USB is attached.

ClockSelect, 2016-10-19:
No news good news

closed 2016-10-19 by ClockSelect

fonix232, 2016-10-22:
Sorry for not replying, haven't got the notification. It is indeed a solution, however sleep mode would be preferred to turn on only if there's no charging.

Seeing how you already made a commit to fix this, I'll see if it fixes my other problem I noticed lately - namely, the batteries are not charged after a time (presumably after entering sleep mode?). It simply shows charging, and voltage stays the same.

fonix232, 2016-10-22:
Okay, it indeed fixed the charging issue I was experiencing. Instead of charging 1-2% in 4 hours, I charged from 35% to 99% in two (using two LG HG2)

ClockSelect, 2016-10-22:
Great!