ClockSelect/myevic #86: Decrease highest step speed

fonix232, 2016-10-17:
The highest step speed seems to be WAY too fast compared to the slower steps. It would be nice if this could be changed, or decreased by default.

Nestorfish, 2016-10-17:
I disabled the last speed, it was way too fast for me too.

ClockSelect, 2016-10-18:
With the options to increase by + or -1 Watt, this shouldn't be a concern any more.
The terminal velocity is reach after more than 5 seconds pressing the button. This just means "go all the way up".

ClockSelect, 2016-10-18:
Problem arise when dealing with high-power boxes. When you have to go through 30 to 75, it's rather easy to do with a reasinable speed. But soon, we'll have to go from say 30 to 250, and it's no longer easy to deal with. We don't know where people want to stop when they start to press the button. There is no universal solution. Ideas welcome.

t2dYuri, 2016-10-18:
Maybe some initial check with condition like:
if([max.wattage] <= 80) {
  use [slow-speed-rotation method];
} else {
  use [high-speed-rotation method];
}

and it would be nice if the user could choose step by 0.5 Watt (besides 0.1 & 1.0) - it's for low-wattage-vapers, who runs their boxes at 10-30 watts, where +-0.5W can be felt?
Thank you.

fonix232, 2016-10-18:
I understand the need for higher speeds, however the overall menu stepping speeds are not distributed in a balanced way.

Say I want to get from the default 20W to 80-90W. I start pressing +, number starts to roll, and in 5 seconds it reaches approx. 35W. After that it begins to spin out of control and needs VERY precise timing to hit near the wanted value.

So either increase the speed of the second fastest stepping, or decrease the speed of the fastest, which would allow for more precision.

In my experience manufacturers got this bit quite right in the stock firmwares, so maybe a device-specific stepping speed implementation would help? Say, a simple switch:

```
switch(currentDevice) {
    case CUBOID:
        speed1 = CUBOID.speed1;
        speed2 = CUBOID.speed2;
        speed3 = CUBOID.speed3;
        break;
    case RX75:
    case [...]:
        break;
    default:
        // set default 
        break;
```

ClockSelect, 2016-10-19:
I don't want to make complicated things.
The way OFW handle speeds is bad, obviously; I prefear the eleaf way.
I'll try things.

dabugen, 2016-10-19:
Honestly, I like the speeds how they are right now in your firmware.

Nestorfish, 2016-10-19:
All this may be due to the way the speed change, and not the maximum speed itself.
I tried a progressive ramp:
https://github.com/Nestorfish/myevic/commit/4575eb44b8b63fc4d093970f879b4063ac974e35
It prevents speed jumps accross sections, reaching the "1W" step at 1W/s after 4.6 seconds, then progressively going to 10W/s for another 3.6 seconds. Maybe this last step can be made a little shorter.

ClockSelect, 2016-10-20:
@Nestorfish: seems fine like this - need some practice to get used to it but eh.

closed 2016-10-26 by ClockSelect