ClockSelect/myevic #87: Ptogrammable Upper Temp Limit

Pipes59, 2016-10-17:
I have had a couple instances of the temperature running up to max while in my pocket. Would be nice to reduce the limit down closer to what one generally uses to avoid such mishaps.
Thank you all for your efforts and consideration.

ClockSelect, 2016-10-18:
The box cannot guess it's in your pocket.
It can make make no difference between your fingers and your keys (or wathever).
If things you have in your pocket make the box prone to respond to buttons, switch off the box.
Won't fix.

Pipes59, 2016-10-18:
Yes, I understand the just turn it off solution.
I think I should have mentioned the other benefit. I have trouble seeing the display without my glasses.
If by pushing up a few clicks I'd know it is at it's max. Let's say I generally like 350F and I had upper limit set to 400F. Then all I would have to do is click up a numerous times and then click down 5 times and I know it's at 350F. Handy while driving too. Basically, makes temp setting possible by counting.
Sorry I didn't add this initially.

MarkyAD, 2016-10-18:
press and hold both + and - until it shows "key lock". it can still fire but power and temperature cannot be changed.

ClockSelect, 2016-10-18:
Agree, key lock is done for that.

closed 2016-10-19 by ClockSelect