ClockSelect/myevic #88: Where to donate?

dabugen, 2016-10-19:
Now honestly, you guys did put so much work into this already and are still adding new features + new releases daily which I am download almost each day as well, yet I can´t see where I could donate for that great stuff... Please let me know:)

ClockSelect, 2016-10-19:
I didn't expect that, so didn't do anything this way.

closed 2016-10-21 by ClockSelect

fonix232, 2016-10-22:
I double this. I know jacksh*t about C/C++ and working with hardware level stuff, but I use these awesome tweaks you did (not to mention the months' work with reversing the stock firmware!), so the least I can do is give you some beer money. I know you're not doing this for the cash, but in a way, many would like to show appreciation 😄