ClockSelect/myevic #89: Deviation in non-temp resistance / inconsistent vape

MaRCoMLXXV, 2016-10-19:
First of all, thanks for all your time and effort you've put in this custom firmware. I really love it.
Just uploaded the current release onto one of my Presa TC 75W's and noticed something strange. I currently have a 1.41 Ohm Kanthal A1 build but the latest release shows it as 1.51 Ohm. My Ohmmeter, several other devices and even the Presa with stock firmware, reads the resistance correctly. I haven't changed any settings. I did check shunt resistance to be sure, but it's at 100%. When firing the correct resistance is shown at the upper line with the protection progress bar. I changed the display to display real time resistance and it's all over the place. I'm aware it's always jumping somewhat, but this seems quite extreme. First time firing, with a higher wattage than usual because of the deviation in resistance, gives a decent vape. However, every following time I press fire, it's more like hit and miss. Could it be that the (unjustified) huge fluctuations in the coils resistance are causing this? As mentioned above, tested on different devices, even checked my connections, but all is just fine, so it seems that the only change has been the new firmware? I'm back to stock right now, as I accidentally deleted the previous version and I can't find a way to download the previous version to revert to that.

Thanks again, looking forward to your reply.

ClockSelect, 2016-10-19:
The shunt adjustement is not a percentage; it's an absolute value in milliOhms (but it's 100mOhms on the Presa 75 so it's OK). A bad shunt value would read a wrong resistance, but it would be systematic, it wouldn't jump all over the place.

All previous versions can be found under the "Commits" section; you can browse the history an revert to any version you like. Can you remember when you got the former version?

My resistance readings are working fine on my VTCs and I can't see where I could have messed up something. Do you have the same behavior with all resistances?

MaRCoMLXXV, 2016-10-19:
Thanks for your reply and the clarification about the shunt adjustment. 

I'm not sure, but I think the previous version was from about 2-3 weeks ago, the last build prior to when the shunt correction was added. 

As for your last question, the two atomizers I currently use have exactly the same resistance, and both have exactly the same issue. I will try some other resistances tomorrow, as well as other wires, see if that makes a difference. I simply can't understand why the correct resistance is shown while firing in power mode, while it keeps showing a higher resistance on the third line before switching off the display.

ClockSelect, 2016-10-20:
@Daveee10 : You have a Presa iirc; do you experiment the same thing?

Daveee10, 2016-10-20:
No. For me no problem with resistance measurement.
Just a little bit difference beetween my VTC Mini and Presa... Only +-0,03 (Maximum)

MaRCoMLXXV, 2016-10-20:
@Daveee10 Are you using Kanthal or a TC wire?

Daveee10, 2016-10-20:
I'm using SS316, SS304, Kanthal A.
If you want I can test with SS410, SS430, Dicodes Resistherm too...

ClockSelect, 2016-10-22:
I have something with thin wires, that are measured right at box startup and a bit high later, but it's not Kanthal, it's more SS prone to heat a little between probes. I've no problem with Kanthal, but they're low gauge. Can you see a difference with thickness af the coil? Like between 26 and 30?

Daveee10, 2016-10-22:
Hmm... I updated to last firmwares..
From this commit - [#86] Tweak key repeat speed the resistance not good...
Before firing shows: 1,31Ohm but when I'm firing the resistance shows 1,02 and this is the good rez...

ClockSelect, 2016-10-22:
It seems it was related to multi-cell devices integration.
Give a try to the suggested fix.

MaRCoMLXXV, 2016-10-23:
Just tried the new build, looks like it's fixed. Tried both Kanthal and SS, resistance is displayed correctly. Thanks!

Just a small heads up, unrelated but nevertheless, the link to MicroSur's FWUpdater in readme.md is broken. Found the correct link here: http://www.ecigtalk.ru/forum/f16/t101158.html

closed 2016-10-23 by MaRCoMLXXV

ClockSelect, 2016-10-23:
I fixed the link to FWUpdater already; it work now. Thank you.